ronengr/hass_nuki_bt #5: How i can debug ?

Balooforever, 2023-05-04:
### System Health details

Home Assistant in Unraid VM. Bluetooth Key UGREEN & Shelly v2 with Proxi

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ludeeus/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hi,

Thank you very much for your work, I've been waiting for it for a long time!

Small question, how can we debug this integration ?

Nothing appears in the discovery (Nuki Smart Lock 3 with Firmware 3.3.5), I use a bluetooth key & a shelly as a bluetooth proxy.

Thank you for your help

### Reproduction steps

No Nuki Smart Lock 3 discovery

### Debug logs

```text
Nothing about bluetooth in the log
```

ronengr, 2023-05-04:
As far as i know, Shelly currently only supports passive bluetooth proxy. It only supports forwarding the ibeacon and does not support BLE connections (which is required to communicate with Nuki Lock)

Balooforever, 2023-05-04:
Oh, I'm looking tonight to move the Nuki Lock and see if everything works :)

Thanks

Balooforever, 2023-05-05:
Everything works via the bluetooth key. Indeed the Shelly only works with iBeacon, but does not relay (located at 3m :( )

I just have to move the lock :)

closed 2023-05-05 by Balooforever