ronengr/hass_nuki_bt #10: Add support for Nuki Opener

bliepp, 2023-08-16:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Right now (as far as I understand the source code) only Nuki SmartLocks are supported. It would be nice to have the ability to use Nuki Openers with this intgration as well.

### Describe the solution you'd like

The Nuki Opener API seems to be implemented in PyNukiBT as far as I understand (at least there are the correct API constants available, but the actual implementation might have some flaws as the NukiDevice class calls some LockActions that are not present in the NukiOpenerConst class - I'm not sure about that, though, as I was unable to test it).

### Describe alternatives you've considered

Buying a Bridge or using a Nuki Hub or the ESPHome extension, but I want to avoid extra hardware.

### Additional context

---

bliepp, 2023-08-16:
I did a quick dive into the source and I think it's basically these methods that need to be modified:

https://github.com/ronengr/hass_nuki_bt/blob/6270b2729b9abb3183e7000dbc92518f23614732/custom_components/hass_nuki_bt/lock.py#L63-L73

The provided constants should be `NukiOpenerConst.LockAction.<###>` where `<###>` is whatever the lock action for a Opener should be (NukiHub and the official integration that needs the bridge use the lock/unlock mechanism to toggle Ring To Open or Continuous Mode as far as I know, so (DE)ACTIVATE_CM would make sense for (un)lock and ELECTRIC_STRIKE_ACTUATION for opening).

ronengr, 2023-08-16:
I would love to add support for Nuki Opener. Unfortunately I don't have one and am not really familiar with it and it's functionality.
I did my best to keep the support from it in PyNukiBT, but as i have no way to test it i'm not sure if it really works.

If you have an opener and are interested in adding the necessary code changes, I would gladly add them.
Please let me know if there is any way I can help.

bliepp, 2023-08-16:
Thanks for the reply. I'm already digging through the code and try my best, but I can hardly tell if I will succeed as I have absolutely no experience in developing integrations for Home Assistant. Guess I'll try to make the features work step by step and we'll see how far I come. What' the best way to ask further (and probably more detailed) questions? Do you prefer general questions being asked here or would you prefer activating Github Discussions for that case?

ronengr, 2023-08-17:
I also don't have any experience in developing HA integrations. This is my first one :-)
Regarding further questions - whatever you prefer. I've activated Github Discussions (or at least i think i did), but this is not a very active repo so you can continue asking here.

One tip i can already give you is to first verify the pyNukiBT part is working. The way i did it was using my modified version of RaspiNukiBridge that i changed to use pyNukiBT. This way you can let RaspiNukiBridge manage and save the lock credentials and use curl to send/receive commands. 
I tried to keep things generic so they wold work with the opener, but as i said i had no way to test so i have no idea if it actually works.

Once pyNukiBT works with the opener, changing the HA integration should be the easy part.

bliepp, 2023-08-18:
Thanks for the advice. That will probably save me hours of debugging 😅

From a first look it seems okay, but I already spotted some things that are not written generic enough

bliepp, 2023-08-25:
Okay, so I couldn't make your modified version of RaspiNukiBridge work due to some async/await errors, so I'm currently writing my own test environment for pyNukiBT.  It looks promising so far, as I was able to pair and pyNukiBT seems to recognize my Opener as an Nuki Opener device, but until now trying to communicate with it (e.g. retrieving its config) leads to a timeout error. Not sure if it's a package error or just me being confused.

ronengr, 2023-08-26:
Can you share the error log you are getting with RaspiNukiBridge?

Also can you share the log of pyNukiBT's parsing issue?

mayabe I can understand the issue from the logs

bliepp, 2023-11-10:
Sorry for not answering for a while. I got caught up with uni (exams and stuff).

@bklaesener If I remember correctly this is the same error I got. It basically says that pairing failed. Actually pairing an Opener isn't currently possible with this project, so making sure the underlying package works is the first step. To further test the PyNukiBt package against the Opener I wrote a utility script that verifies pairing an Opener is technically possible, but there are some errors when actually communicating (e.g. requesting the current config results in some error). Expect my script to be posted here soon.

bliepp, 2023-11-10:
Here's my test environment. It's basically the relevant parts of this project copied and pasted into one script. The requirements are PyNukiBt and nacl.

To use it, set your opener into pairing mode by pressing the button for 6 seconds until the LED is blinking. Then run the script. It will search for BLE devices called `Nuki_Opener_*`, pair with the first one it finds and print the device's connection data and type on success. It then connects to it, attempts to update the config to gain info about the battery percentage. This is where my script fails saying it cannot access the opener (however, pairing and initializing a conenction works). When I'm back home again I will retry it and post the script's output here, but I guess it's some error due to lack of access right.

Oh, I had to upload tge pytho nscript as a `*.txt` as GitHub doesnt allow attachments to be a python file.
[Download testenv.txt](https://github.com/ronengr/hass_nuki_bt/files/13318735/testenv.txt)

bliepp, 2023-11-12:
Okay, second attempt.

Here's my test environment: https://github.com/ronengr/hass_nuki_bt/files/13328487/testenv.zip
It's basically a minimal example of this project by copying and pasting the relevant parts into one script. Make sure you have PyNukiBt installed.

To use it, activate pairing mode on your Opener by pressing the button for 6 seconds until the LED is blinking. Then run the script. It will search for BLE devices called `Nuki_Opener_<gibberish>`, pair with the first one it finds and print the device's connection data and type on success. It then connects to it, attempts to update the config to gain info about the battery percentage. It works until it gets to actually pair with the Opener and I have no idea why. In the attached zip file there's also a `log.txt`file with the scripts output on my machine.

ronengr, 2023-11-26:
Looking at the logs it seems to try to connect using the Lock BLE char.
my guess, the code that should recognize if it is a lock or an opener (and set the self._const accordingly) is not identifying the opener correctly.

Looking at the code, the way it identifies Nuki type is by checking for  NukiOpenerConst.BLE_PAIRING_CHAR in the ble Characteristics and if it is not there it assumes its a lock.

maybe the BLE IDs for the opener i have are wrong?

could you try again with debug logs enabled?
there is a debug log that prints the services and all the Characteristics of the device before trying to connect. this way we can find out if we have the correct char id or not

nico1510, 2024-03-01:
Hey @ronengr I'm not the original poster but ran into the same issue. The BLE IDs for the opener *seem* to be correct but I'm not experienced with bluetooth at all. To me the debug logs seem to indicate that connection is successful but parsing the messages from the opener fails. Here are my debug logs please let me know if you have any pointers for me :slightly_smiling_face: 

```
vscode ➜ /workspaces/hass_nuki_bt (main) $ ./scripts/develop 
2024-03-01 07:50:46.228 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hass_nuki_bt which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-03-01 07:50:59.031 WARNING (MainThread) [asyncio] Executing <Task pending name='setup component analytics' coro=<_async_setup_component() running at /home/vscode/.local/lib/python3.10/site-packages/homeassistant/setup.py:288> wait_for=<Task pending name='Storage load core.analytics' coro=<Store._async_load() running at /home/vscode/.local/lib/python3.10/site-packages/homeassistant/helpers/storage.py:124> cb=[set.remove(), Task.task_wakeup()] created at /home/vscode/.local/lib/python3.10/site-packages/homeassistant/core.py:560> cb=[set.remove(), Task.task_wakeup()] created at /home/vscode/.local/lib/python3.10/site-packages/homeassistant/core.py:560> took 0.349 seconds
2024-03-01 07:51:08.084 WARNING (MainThread) [homeassistant.components.debugpy] Listening for remote debug connection on 0.0.0.0:5678
2024-03-01 07:52:22.433 WARNING (MainThread) [asyncio] Executing <Task pending name='setup component ssdp' coro=<_async_setup_component() running at /home/vscode/.local/lib/python3.10/site-packages/homeassistant/setup.py:288> wait_for=<_GatheringFuture pending cb=[Task.task_wakeup()] created at /usr/local/lib/python3.10/asyncio/tasks.py:665> cb=[set.remove(), Task.task_wakeup()] created at /home/vscode/.local/lib/python3.10/site-packages/homeassistant/core.py:560> took 0.120 seconds
2024-03-01 07:52:36.184 WARNING (MainThread) [asyncio] Executing <Task pending name='setup component cloud' coro=<_async_setup_component() running at /home/vscode/.local/lib/python3.10/site-packages/homeassistant/setup.py:288> wait_for=<Task pending name='Storage load cloud' coro=<Store._async_load() running at /home/vscode/.local/lib/python3.10/site-packages/homeassistant/helpers/storage.py:124> cb=[set.remove(), Task.task_wakeup()] created at /home/vscode/.local/lib/python3.10/site-packages/homeassistant/core.py:560> cb=[set.remove(), Task.task_wakeup()] created at /home/vscode/.local/lib/python3.10/site-packages/homeassistant/core.py:560> took 0.754 seconds
2024-03-01 07:52:36.219 ERROR (MainThread) [homeassistant.bootstrap] Error setting up integration bluetooth - received exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pkg_resources/__init__.py", line 3108, in _dep_map
    return self.__dep_map
  File "/usr/local/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2901, in __getattr__
    raise AttributeError(attr)
AttributeError: _DistInfoDistribution__dep_map

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pkg_resources/__init__.py", line 3099, in _parsed_pkg_info
    return self._pkg_info
  File "/usr/local/lib/python3.10/site-packages/pkg_resources/__init__.py", line 2901, in __getattr__
    raise AttributeError(attr)
AttributeError: _pkg_info

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/vscode/.local/lib/python3.10/site-packages/homeassistant/setup.py", line 102, in async_setup_component
    return await task
  File "/home/vscode/.local/lib/python3.10/site-packages/homeassistant/setup.py", line 207, in _async_setup_component
    await async_process_deps_reqs(hass, config, integration)
  File "/home/vscode/.local/lib/python3.10/site-packages/homeassistant/setup.py", line 420, in async_process_deps_reqs
    await requirements.async_get_integration_with_requirements(
  File "/home/vscode/.local/lib/python3.10/site-packages/homeassistant/requirements.py", line 52, in async_get_integration_with_requirements
    return await manager.async_get_integration_with_requirements(domain)
  File "/home/vscode/.local/lib/python3.10/site-packages/homeassistant/requirements.py", line 171, in async_get_integration_with_requirements
    await self._async_process_integration(integration, done)
  File "/home/vscode/.local/lib/python3.10/site-packages/homeassistant/requirements.py", line 186, in _async_process_integration
    await self.async_process_requirements(
  File "/home/vscode/.local/lib/python3.10/site-packages/homeassistant/requirements.py", line 252, in async_process_requirements
    await self._async_process_requirements(name, missing)
  File "/home/vscode/.local/lib/python3.10/site-packages/homeassistant/requirements.py", line 284, in _async_process_requirements
    installed, failures = await self.hass.async_add_executor_job(
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/vscode/.local/lib/python3.10/site-packages/homeassistant/requirements.py", line 113, in _install_requirements_if_missing
    if pkg_util.is_installed(req) or _install_with_retry(req, kwargs):
[... cut ...]
  File "/usr/local/lib/python3.10/selectors.py", line 469, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
^CTraceback (most recent call last):
  File "/home/vscode/.local/lib/python3.10/site-packages/debugpy/_vendored/pydevd/pydevd.py", line 2299, in __wait_for_threads_to_finish
    time.sleep(1 / 10.)
  File "/home/vscode/.local/lib/python3.10/site-packages/homeassistant/util/async_.py", line 167, in protected_loop_func
    return func(*args, **kwargs)
KeyboardInterrupt
^C
vscode ➜ /workspaces/hass_nuki_bt (main) $ 
```

ronengr, 2024-03-31:
@nico1510 indeed it seems there is an error parsing the message.
It seems that the decryption is incorrect for some reason.

You can see that the unencrypted part of the message seems to be OK (length make sense and the auth-id is consistent), but in the encrypted part you get a lot of strange things: the auth-id is always different, the command ID is different and doesn't correspond to any known command, etc.

(you can find the message format here: https://developer.nuki.io/page/nuki-opener-api-1/7)

the strange thing is that pairing seems to be successful, so I'm not sure why the decryption fails.
maybe the message format for opener is different that the lock (and different than the docs)

you can try to print the unencrypted message by changing the function `_parse_message()`
simply save and print the value of `self._decrypt_message(data)` before passing it to `self._const.NukiMessage.parse()`
hopefully from there you can find if the message is the same as in the documentation or not. try looking for the authID (same 4 bytes that are 24:27 in the unencrypted message part), maybe there is some offset.

another strange thing i noticed is that the crc is always the same - 2024, which makes me believe the decryption maybe did succeed but the issue is with the message format (2024 could be the year, which is part of the expected message)

011V32, 2024-06-09:
Does this already work?
The Opener gets discovered...
[image]

ronengr, 2024-06-09:
@011V32 
no idea. I don't have an opener so i have no way of testing.
Latest changes could have solved the opener issues.
Only way to really know is to try :-)

If it still doesn't work please send me the logs and i will try to fix.

bklaesener, 2024-07-18:
I tried, did not work. Here are the relevant log entries:

```
Logger: pyNukiBT.nuki
Source: /usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py:350
First occurred: 14:46:23 (1 occurrences)
Last logged: 14:46:23

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 346, in _send_command
    msg = await self._notify_future
          ^^^^^^^^^^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 345, in _send_command
    async with async_timeout.timeout(self.command_response_timeout):
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
TimeoutError

Logger: pyNukiBT.nuki
Source: runner.py:190
First occurred: 14:46:02 (2 occurrences)
Last logged: 14:46:29

Got unexpected message length for command KEYTURNER_STATES. Got 8 unknown bytes with value: b'\xe8\x07\x07\x12\x08\x0e\x1d\x00'
Error K_ERROR_BAD_NONCE, command REQUEST_CONFIG

Logger: homeassistant.config_entries
Source: config_entries.py:586
First occurred: 14:46:29 (1 occurrences)
Last logged: 14:46:29

Error setting up entry Nuki Opener for hass_nuki_bt
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 586, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 90, in async_setup_entry
    if not await coordinator.async_wait_ready():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 155, in async_wait_ready
    await self._async_update()
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 112, in _async_update
    await self.device.update_state()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 434, in update_state
    await self.update_config()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 489, in update_config
    msg = await self._send_encrtypted_command(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 196, in _send_encrtypted_command
    return await self._send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 346, in _send_command
    msg = await self._notify_future
          ^^^^^^^^^^^^^^^^^^^^^^^^^
pyNukiBT.const.NukiErrorException
```

Randolf22, 2024-08-21:
I also have the same issue:
Logger: pyNukiBT.nuki
Quelle: /usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py:350
Erstmals aufgetreten: 11:10:59 (3 Vorkommnisse)
Zuletzt protokolliert: 11:11:41

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 346, in _send_command
    msg = await self._notify_future
          ^^^^^^^^^^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 345, in _send_command
    async with async_timeout.timeout(self.command_response_timeout):
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
TimeoutError

Randolf22, 2024-09-04:
No ideas?

ronengr, 2024-11-11:
@Randolf22 @bklaesener 
can you enable debug logs and send more detailed logs?

It looks like getting the state works, at least partially. but i cannot be sure without full logs
seems there is some error getting the config, but again without full logs i cannot debug this.

0SkillAllLuck, 2025-01-30:
Started to try and add support for the opener, got most things working now. Lock switches on/off the RTO mode, also added buttons for Continous mode on/off. Battery percentage doesn't seem to be available sadly so just made it report 0% if the battery is critical and 100% if its not.

Needs some changes in pyNukiBT that I created a [PR](https://github.com/ronengr/pyNukiBT/pull/22) for already. Feel free to dismiss if its not how you would do it :D

[image]

ronengr, 2025-02-15:
@0SkillAllLuck 
thanks for your contribution!
I will review the PR and merge it.
Can you also open a PR for the changes you made in hass_nuki_bt?

0SkillAllLuck, 2025-02-16:
Yeah, you might need to create a new version of pyNukiBT first tho. I changed it to 0.0.18 in the PR as I created that version locally.

berrywhite96, 2025-03-28:
Great work guys!

Am I blind or its currently not logged when someone rings? It looks like the Nuki integration supports it.

0SkillAllLuck, 2025-03-30:
No currently its not. I have not had the time to figure out how to correctly detect when someone rings.
Must be possible as the bridge manages to as well tho.

bliepp, 2025-04-18:
Hey there, man, there's a lot going on here since I checked in the last time.
@0SkillAllLuck thank you very much for the implementation.

So from now on, the Opener should work when installing this integration via HACS, right? Because when I try to setup my Opener via the UI, an empty error dialog pops up. Checking the homeassistant logs, I get this error message:

<details>
  <summary>Click here to open logs</summary>

```
2025-04-18 22:33:42.531 ERROR (MainThread) [aiohttp.server] Error handling request from 192.168.178.137
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 232, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 121, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 340, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 383, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, user_input["next_step_id"], None
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 490, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/config_flow.py", line 158, in async_step_pair
    await device.connect()
  File "/usr/local/lib/python3.13/site-packages/pyNukiBT/nuki.py", line 394, in connect
    await self._client.connect()
  File "/usr/local/lib/python3.13/site-packages/habluetooth/wrappers.py", line 303, in connect
    wrapped_backend = self._async_get_best_available_backend_and_device(manager)
  File "/usr/local/lib/python3.13/site-packages/habluetooth/wrappers.py", line 426, in _async_get_best_available_backend_and_device
    raise BleakError(
    ...<2 lines>...
    )
bleak.exc.BleakError: No backend with an available connection slot that can reach address 54:D2:72:25:9D:3A was found
```

</details>

Initially, it was auto-discovered by HA, but after the first failed attempt the auto-discovery was gone and I had to try again with the manual setup.

0SkillAllLuck, 2025-04-19:
Looks a bit more like a general BLE issue to me than from the pyNukiBT library. Maybe your BLE adapter had some issues.

bliepp, 2025-04-19:
> Looks a bit more like a general BLE issue to me than from the pyNukiBT library. Maybe your BLE adapter had some issues.

Digging deeper into the issue you might be correct, see #7

Ralf-Th, 2025-06-19:
[image]

i have the opener successfully connected, but which action to issue the open command? nothing happens on device itself

0SkillAllLuck, 2025-06-19:
[image]

In the HomeAssistant app you have the "Open Door" button. That opens the door. The lock itself enabled continuous mode.

Ralf-Th, 2025-06-29:
I still have no function with the opener

ffMathy, 2025-06-29:
@Ralf-Th it works for me just fine. Is your Home Assistant within roughly 6 meters of your Opener? The Bluetooth signal I found is pretty weak.

Ralf-Th, 2025-06-29:
iuse shelly proxies, the next one is 4 meters away

ffMathy, 2025-06-29:
What are your symptoms?

Ralf-Th, 2025-06-29:
No response from the opener

ronengr, 2025-06-29:
@Ralf-Th 
If I remember correctly, the Shelly proxy is only passive and not active. Nuki needs an active connection, so the Shelly proxy will probably not work.

Ralf-Th, 2025-06-30:
so, schould i use app mode instead of gateway?

Ralf-Th, 2025-06-30:
for example, my Jura coffee machine i can handle perfectly

ronengr, 2025-06-30:
You should either bring your HA closer to the Nuki or use esphome proxy.

There is no difference between app and bridge mode. Both require full 2 way communication.

Ralf-Th, 2025-06-30:
i already have an esphome bt proxy, may be to far away. i thought the shellies repeat the signal

Ralf-Th, 2025-07-16:
no chance to use the integration?

metallus42, 2026-09-17:
How about Ring detection? Is it even possible to extract that from the bt api? I know it can be done via nubi web and the official hub