ronengr/hass_nuki_bt #12: Supervisor 2023.08.2 breaks hass_nuki_bt

HaJoSa, 2023-08-22:
### System Health details

## System Information

version | core-2023.8.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.32.1
Stage | running
Available Repositories | 1270
Downloaded Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.08.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 439.4 GB
disk_used | 9.1 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | deCONZ (6.20.0), RaspberryMatic CCU (3.69.7.20230626), Home Assistant Google Drive Backup (0.111.1), Advanced SSH & Web Terminal (15.0.7), File editor (5.6.0), Samba share (10.0.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 2
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 12. August 2023 um 19:13
-- | --
current_recorder_run | 21. August 2023 um 21:55
estimated_db_size | 281.29 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

After installing Supervisor 2023.08.2 from 2023.08.1 nearly all entities from hass_nuki_bt are gone.
After restoring my latest backup, hass_nuki_bt works as it should.

### Reproduction steps

1. install Supervisor 2023.08.2
2.
3.
...

### Debug logs

```text
Sorry, no logger
```

ronengr, 2023-08-23:
From what I can see, the latest Supervisor is 2023.08.1. Supervisor version 2023.08.2 is not out yet

I have Home Assistant 2023.8.3 with Supervisor 2023.08.1, and for me everything works

HaJoSa, 2023-08-24:
You are right, as of today latest Supervisor version is 2023.08.01.
But last Monday evening I got version 2023.08.02 ready for installation, maybe accidently.