ronengr/hass_nuki_bt #13: Not Open Door control

alfredocdmiranda, 2023-08-23:
### System Health details

## System Information

version | core-2023.8.2
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.4.11-200.fc38.x86_64
arch | x86_64
timezone | Europe/Lisbon
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4970
Installed Version | 1.32.1
Stage | running
Available Repositories | 1267
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | August 13, 2023 at 10:26 AM
-- | --
current_recorder_run | August 23, 2023 at 5:22 PM
estimated_db_size | 22.70 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..
[... 101 lines cut ...]
        3
        6
        9
    hardware_revision = ListContainer: 
        6
        11
    homekit_status = 1
    timezone_id = (enum) EUROPE_LONDON 40
    undocumented = 4
    undocumented2 = 0
    has_keypad_v2 = 0
2023-08-23 17:22:37.990 INFO (MainThread) [pyNukiBT.nuki] request 1 log entries, start=0
2023-08-23 17:22:37.990 INFO (MainThread) [pyNukiBT.nuki] sending encrypted command REQUEST_DATA
2023-08-23 17:22:37.990 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1
2023-08-23 17:22:37.990 INFO (MainThread) [pyNukiBT.nuki] Connected
2023-08-23 17:22:37.990 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2023-08-23 17:22:38.091 DEBUG (MainThread) [pyNukiBT.nuki] Notification handler: a92ee202-5501-11e4-916c-0800200c9a66 (Handle: 32): Unknown, data: bytearray(b"LV\xe3)\xa3\x95l\x16\xf1\xdf\xd6\xc5.tk5 \xc0\x8b\x06#\xef\xefr\x01\x00\xbfu8\x00u3\x18\x8dlD\xfd\x17\xbe\\\ns\xf0\x05HB\xb3\xd8\xe4V\xe4\xab\xb7\x95\x7f\'\x18R\xb42\x15[\xb2\x0c\xb9\x17\xc7qO\xc7\xd4y\r\xfdX\x1a7m`\xc5\x92\x87\x89\xdf\xc5\x8b")
2023-08-23 17:22:38.096 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1
2023-08-23 17:22:38.097 INFO (MainThread) [pyNukiBT.nuki] sending encrypted command REQUEST_LOG_ENTRIES
2023-08-23 17:22:38.097 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1
2023-08-23 17:22:38.097 INFO (MainThread) [pyNukiBT.nuki] Connected
2023-08-23 17:22:38.097 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2023-08-23 17:22:38.197 DEBUG (MainThread) [pyNukiBT.nuki] Notification handler: a92ee202-5501-11e4-916c-0800200c9a66 (Handle: 32): Unknown, data: bytearray(b'I<}=*\xd8\x13\xe3\x93\xcdSv\xb4M#i\xe1!2n\x0e\x82\xb1B\x01\x00\xbfuM\x00\xddF\xac\x02\x96k\x90[,\xb0E\xf2_6J\x07\xe49\x9f"Gp\x97v\xcemr\x9cwS\xd8b\xc6\xb3FN\x0e#%pBX\x11`7\xa1\xde\x16\xecQ5T\xdd\xb5_\xec\xcd\xf3\x88q\xb5\x8av\xbe]3\xa84\xcf\x17\x1e\xa5h\n\xf0\x81`')
2023-08-23 17:22:38.200 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1
```

ronengr, 2023-08-23:
If you click on the "Lock" control the HA control for locks will open, where you have an option to "Open" (or unlatch) the door (the small round button in the photo attached)
[image]

alfredocdmiranda, 2023-08-24:
Hey, thanks a lot. Sorry for this newbie question. It worked perfectly.

closed 2023-08-24 by alfredocdmiranda