ronengr/hass_nuki_bt #17: No Entities on Nuki

pelegw, 2023-09-23:
### System Health details

## System Information

version | core-2023.9.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | Asia/Jerusalem
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4993
Installed Version | 1.32.1
Stage | running
Available Repositories | 1289
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | October 9, 2023 at 3:00 AM
relayer_connected | true
relayer_region | ap-southeast-1
remote_enabled | false
remote_connected | false
alexa_enabled | true
google_enabled | true
remote_server | ap-southeast-1-0.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.09.2
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 28.5 GB
disk_used | 6.3 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Advanced SSH & Web Terminal (15.0.8), Network UPS Tools (0.12.1), Amazon S3 Backup (1.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | September 13, 2023 at 12:54 AM
-- | --
current_recorder_run | September 24, 2023 at 12:18 AM
estimated_db_size | 19.82 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

When adding a Nuki lock to HA, it is found and paired in bridge mode, however no data is received and no entities are seen. The following can be found in the HA logs:

> Error setting up entry Door for hass_nuki_bt
Error setting up entry Nuki_3637497C for hass_nuki_bt
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 79, in async_setup_entry
    if not await coordinator.async_wait_ready():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 148, in async_wait_ready
    await self._async_update()
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 108, in _async_update
    await self.device.update_state()
  File "/usr/local/lib/python3.11/site-packages/pyNukiBT/nuki.py", line 424, in update_state
    self._const.NukiCommand.REQUEST_DATA,
    ^^^^^^^^^^^
AttributeError: 'NukiDevice' object has no attribute '_const'

### Reproduction steps

1.Nuki is detected by HA and pairing starts
2.Long press on Nuki for pairing mode
3.Pairing Finished
4.No entities shown for Nuki

### Debug logs

```text
2023-09-24 00:17:57.060 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hass_nuki_bt which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-24 00:17:57.062 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-09-24 00:18:16.122 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Door for hass_nuki_bt
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
result = await component.async_setup_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/hass_nuki_bt/__init__.py", line 79, in async_setup_entry
if not await coordinator.async_wait_ready():
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/hass_nuki_bt/coordinator.py", line 148, in async_wait_ready
await self._async_update()
File "/config/custom_components/hass_nuki_bt/coordinator.py", line 108, in _async_update
await self.device.update_state()
File "/usr/local/lib/python3.11/site-packages/pyNukiBT/nuki.py", line 424, in update_state
self._const.NukiCommand.REQUEST_DATA,
^^^^^^^^^^^
AttributeError: 'NukiDevice' object has no attribute '_const'
```

pelegw, 2023-09-24:
This is suddenly working now - the only thing I did was synchronize time on the lock. Weird.

pelegw, 2023-09-24:
Never mind, it's back to the same error again. Really weird.

> This is suddenly working now - the only thing I did was synchronize time on the lock. Weird.

ronengr, 2023-10-17:
This is an interesting error.
I will look into it

ronengr, 2023-10-17:
I have released a new version that should fix this issue.
can you please check?

pelegw, 2023-10-17:
Not really, I’ve been called up for reserve service :/

ronengr, 2023-10-17:
@pelegw keep safe

melegio, 2024-01-11:
I have the same error.
How I can solve it?

fabermundi, 2024-02-06:
Hello,

Unfortunately I also have to same error. It seems that a function signature has change (number of parameters).

ronengr, 2024-04-05:
@fabermundi @melegio 
can you please check if it still happens in latest version?
I've did some changes in the code that might fix this issue.

melegio, 2024-04-06:
Sorry, The issue is still there.

ronengr, 2024-04-06:
@melegio 
are you getting the same error?
how far is the lock from your BT device?

from the looks of the last logs you attached it seems like a connection issue - try putting the BT closer to the lock.

fabermundi, 2024-04-08:
In addition to the before mentioned exception I also get:

  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 356, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/hass_nuki_bt/button.py", line 46, in async_setup_entry
    async_add_entities([NukiButton(coordinator, btn) for btn in BUTTON_TYPES])
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/button.py", line 58, in __init__
    super().__init__(coordinator)
  File "/config/custom_components/hass_nuki_bt/entity.py", line 42, in __init__
    str(x) for x in coordinator.device.config["hardware_revision"]
                    ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'hardware_revision'

For the entities "button", "sensor", "lock" and "binary_sensor".

An the exception

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 366, in _send_command
    msg = await self._notify_future
          ^^^^^^^^^^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/active_update_coordinator.py", line 131, in _async_poll
    self.data = await self._async_poll_data(self._last_service_info)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/active_update_coordinator.py", line 124, in _async_poll_data
    return await self._poll_method(last_service_info)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 110, in _async_update
    await self.device.update_state()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 428, in update_state
    msg = await self._send_encrtypted_command(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 193, in _send_encrtypted_command
    return await self._send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 365, in _send_command
    async with async_timeout.timeout(self.command_response_timeout):
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
TimeoutError

Very strange, that I get a TimeoutError because as mentioned the Raspberry is just next to the Nuki.

melegio, 2024-04-11:
My BT is 2mt far from the nuki. This is the error that I get.
Logger: homeassistant.components.button
Source: helpers/entity_platform.py:356
integration: Button (documentation, issues)
First occurred: April 9, 2024 at 6:25:51 PM (4 occurrences)
Last logged: 7:04:06 PM

Error while setting up hass_nuki_bt platform for button
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 356, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/hass_nuki_bt/button.py", line 46, in async_setup_entry
    async_add_entities([NukiButton(coordinator, btn) for btn in BUTTON_TYPES])
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/button.py", line 58, in __init__
    super().__init__(coordinator)
  File "/config/custom_components/hass_nuki_bt/entity.py", line 42, in __init__
    str(x) for x in coordinator.device.config["hardware_revision"]
                    ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'hardware_revision'

ronengr, 2024-04-11:
@fabermundi 
these are strange error you are getting about the hardware_revision
can you give me some more debug logs? it's hard to say from the error why there is no hardware revision in the config.
there should be a debug print of the driver getting the config info from the lock, I'm interested to see if the hw revision is just missing there or what the issue is.

for now, you can add a small patch locally to avoid an exception in this case, but i would prefer finding the root-cause and not just ignoring the isue.

`diff --git a/custom_components/hass_nuki_bt/entity.py b/custom_components/hass_nuki_bt/entity.py
index 601b173..7c40ade 100644
--- a/custom_components/hass_nuki_bt/entity.py
+++ b/custom_components/hass_nuki_bt/entity.py
@@ -39,10 +39,10 @@ class NukiEntity(PassiveBluetoothCoordinatorEntity[NukiDataUpdateCoordinator]):
             model=coordinator.device.device_type,
             name=coordinator.device_name,
             hw_version=".".join(
-                str(x) for x in coordinator.device.config["hardware_revision"]
+                str(x) for x in coordinator.device.config.get("hardware_revision",[])
             ),
             sw_version=".".join(
-                str(x) for x in coordinator.device.config["firmware_version"]
+                str(x) for x in coordinator.device.config.get("firmware_version",[])
             ),
         )
`

ronengr, 2024-04-11:
@melegio 
without context it is hard to debug.
can you add the full debug log so i know in what exact flow the error occurred?

Terabyte2000, 2024-04-21:
@ronengr 

I get the same error with hardware_revision

What do you need to find the failure?

And how can I add your patch you posted before?

Thx

ronengr, 2024-04-21:
@melegio @fabermundi 
can you try with latest version?
Seems the config msg is formatted differently in nuki 4, so i've added ability to ignore config if it is not needed.

melegio, 2024-04-21:
Now it works!!!Thank youGiovanni

closed 2024-04-21 by ronengr