ronengr/hass_nuki_bt #19: Nuki Smart Lock v.3

Mafin-dev, 2023-09-30:
### System Health details

Sadly no logging implemented. After trying to pair my lock I get a "Error" message. Thats it. Nothing with log-debug enabled. 

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Sadly no logging implemented. After trying to pair my lock I get a "Error" message. Thats it. Nothing with log-debug enabled. 

### Reproduction steps

1. automatically pair selected
2. "Error"

### Debug logs

```text
- nothing -
```

Mafin-dev, 2023-10-17:
Thanks for the reply. So the Firmware could be the problem? I don't see any requirements about the firmware version, App Version eg. 
Is it possible to downgrade the Firmware-Version? Could a hardreset solve the issue? I'll try..

oferpaster, 2023-10-17:
> Thanks for the reply. So the Firmware could be the problem? I don't see any requirements about the firmware version, App Version eg. Is it possible to downgrade the Firmware-Version? Could a hardreset solve the issue? I'll try..

My Firmware version is 3.7.7

ronengr, 2023-10-17:
this is strange. I'm also on Firmware version is 3.7.7 and everything is working fine for me.

Do you have any logs of the error?

Mafin-dev, 2023-10-17:
No sadly there are no logs. 
It never worked for me but I always was on FW 3.7.7. 
I'll try to reset my nuki later this evening.

oferpaster, 2023-10-17:
I have logs when trying to connect as App and not Bridge(On Bridge there is nothing):

`2023-10-17 20:18:33.963 INFO (MainThread) [custom_components.hass_nuki_bt] Discovered bluetooth device: {'name': 'Nuki_3163DE74', 'address': '54:D2:72:63:XX:XX', 'rssi': -74, 'manufacturer_data': {76: b'\x02\x15\xa9.\xe2\x00U\x01\x11\xe4\x91l\x08\x00 \x0c\x9af1c\xdet\xc9'}, 'service_data': {'a92ee100-5501-11e4-916c-0800200c9a66': b'1c\xdet'}, 'service_uuids': ['0000003e-0000-1000-8000-0026bb765291', '00000044-0000-1000-8000-0026bb765291', '00000045-0000-1000-8000-0026bb765291', '00000055-0000-1000-8000-0026bb765291', '00000096-0000-1000-8000-0026bb765291', '000000a2-0000-1000-8000-0026bb765291', '00001800-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', 'a92ee100-5501-11e4-916c-0800200c9a66', 'a92ee200-5501-11e4-916c-0800200c9a66'], 'source': '00:2A:7A:00:02:20', 'advertisement': AdvertisementData(local_name='Nuki_XXXXX', manufacturer_data={76: b'\x02\x15\xa9.\xe2\x00U\x01\x11\xe4\x91l\x08\x00 \x0c\x9af1c\xdet\xc9'}, service_data={'a92ee100-5501-11e4-916c-0800200c9a66': b'1c\xdet'}, service_uuids=['0000003e-0000-1000-8000-0026bb765291', '00000044-0000-1000-8000-0026bb765291', '00000045-0000-1000-8000-0026bb765291', '00000055-0000-1000-8000-0026bb765291', '00000096-0000-1000-8000-0026bb765291', '000000a2-0000-1000-8000-0026bb765291', '00001800-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', 'a92ee100-5501-11e4-916c-0800200c9a66', 'a92ee200-5501-11e4-916c-0800200c9a66'], rssi=-74), 'device': BLEDevice(54:D2:72:63:DE:74, Nuki_3163DE74), 'connectable': True, 'time': 7038.852422544}
2023-10-17 20:18:34.060 INFO (MainThread) [pyNukiBT.nuki] Connected
2023-10-17 20:18:34.060 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1
2023-10-17 20:18:34.061 INFO (MainThread) [pyNukiBT.nuki] Connected
2023-10-17 20:18:34.061 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2023-10-17 20:18:34.089 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1
2023-10-17 20:18:34.212 INFO (MainThread) [pyNukiBT.nuki] Nuki 54:D2:72:63:XX:XX public key: 6b96b7da5652fa697a125cf62666f728764d871a68fc31145fb7f85c21b0f950
2023-10-17 20:18:34.212 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1
2023-10-17 20:18:34.213 INFO (MainThread) [pyNukiBT.nuki] Connected
2023-10-17 20:18:34.213 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2023-10-17 20:18:34.239 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1
2023-10-17 20:18:54.241 ERROR (MainThread) [aiohttp.server] Error handling request`

ronengr, 2023-10-17:
This is an interesting log.
It seems that pyNukiBT was successful 
I have no idea what the aiohttp error is. i don't use aiohttp in pyNukiBT/hass_nuki_bt, so this error seems to be unrelated.

oferpaster, 2023-10-17:
> This is an interesting log. It seems that pyNukiBT was successful I have no idea what the aiohttp error is. i don't use aiohttp in pyNukiBT/hass_nuki_bt, so this error seems to be unrelated.

I also got this in the log:
`2023-10-17 22:24:35.844 WARNING (MainThread) [homeassistant.helpers.frame] Detected integration that attempted to call BleakClient with an address instead of a BLEDevice. Please report issue to the custom integration author for hass_nuki_bt using this method at custom_components/hass_nuki_bt/config_flow.py, line 151: ret = await device.pair()`

ronengr, 2023-10-17:
I am aware of this warning.
I'm not sure why HA gives out this warning (not sure why they don't want to use addresses), but it is only a warning and everything should work even if you get the warning.

Let me go over the code again, maybe i can find where the issue is coming from.

at what stage exactly are you getting the error?

oferpaster, 2023-10-17:
> I am aware of this warning. I'm not sure why HA gives out this warning (not sure why they don't want to use addresses), but it is only a warning and everything should work even if you get the [...]

It looks like this error happens only once when tring to connect to nuki after HA restart.

Memphius, 2023-10-17:
Me too. However I used to use the nuki custom add on integration together with a custom add on. It stopped working a few days ago and I couldn't manage to get it work (firmware 3.6 i think it was). 

I uninstalled everything and tried your integration since two days ago. I did update to 3.7 before I tried it out but I keep getting an unspecified error just like the posters.

Anything I can share that might help you find a solution?

Thanks

ronengr, 2023-10-17:
if you have logs that would be very helpful.
at the moment I'm not really sure what the error is or where it happens

ronengr, 2023-10-17:
@oferpaster I've looked again at the logs you sent me, it seems it stops in the middle of paring. (specifically after sending our public key to the nuki)
I'm not sure if we simply didn't get a response from Nuki or if the response was some sort of error.

Can you maybe run it again with loglevel=debug?

oferpaster, 2023-10-18:
Nothing else on logs, I even try to use ESP32 with nuki_hub and its still not working.
I entered to nuki beta and upgrade the firmware to 3.8.1 still not working.

I factory reset nuki - still not working, not hass_nuki_bt and not nuki_hub

Mafin-dev, 2023-10-18:
Factory reset didn't help. Are there any Bluetooth requirements we don't know about? I use a Bluetooth dongle with Bluetooth 5.3. No problems with other Bluetooth devices so far.

oferpaster, 2023-10-18:
I use raspberry pi built in bluetooth and also try usb dongle( supported dongle from Home assistant supported dongles).
I was running nuki from my HA for 1 year already until last week :(

francesco-on-git, 2023-10-18:
I also have the same issue. Could it be related to the operating system version? I'm using Home Assistant OS 11.

oferpaster, 2023-10-23:
How do I change the log level?

guibar, 2023-11-07:
I have the same issue. The lock is detected by HA and I get the option to configure it. But when I do I get an Error with no message whatsoever.
In the logs however, I see this which I am pretty sure is caused by the failure to configure.

```
Logger: aiohttp.server
Source: /usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py:403
First occurred: 6 November 2023 at 19:35:30 (6 occurrences)
Last logged: 22:11:03
Error handling request

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 85, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 31, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 148, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 63, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 177, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 289, in async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 389, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/config_flow.py", line 151, in async_step_pair
    ret = await device.pair()
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pyNukiBT/nuki.py", line 503, in pair
    await self.connect() # connect so we can identify the device type and update self._const accordingly
    ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pyNukiBT/nuki.py", line 386, in connect
    await self._client.connect()
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/wrappers.py", line 273, in connect
    wrapped_backend = self._async_get_best_available_backend_and_device(manager)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/wrappers.py", line 375, in _async_get_best_available_backend_and_device
    raise BleakError(
bleak.exc.BleakError: No backend with an available connection slot that can reach address 54:D2:72:96:44:7C was found

```

ronengr, 2023-11-26:
`bleak.exc.BleakError: No backend with an available connection slot that can reach address 54:D2:72:96:44:7C was found
`
I think this is the main issue. for some reason HA cant find a way to communicate with the Nuki lock.
How are you trying to connect? do you have a BT adapter on the HA PC or are you using a BT proxy?

guibar, 2023-11-27:
> `bleak.exc.BleakError: No backend with an available connection slot that can reach address 54:D2:72:96:44:7C was found ` I think this is the main issue. for some reason HA cant find a way to [...]

I am using the native bluetooth on a raspberry Pi 4. In the end, I got that to work. Not sure what made it work.

ronengr, 2023-11-27:
if it sometimes works and sometimes it doesn't, it could be the distance or other wireless interferences.
Maybe HA is able to detect the lock but when trying to communicate the connection is not stable enough.

Felix313, 2024-01-31:
> if it sometimes works and sometimes it doesn't, it could be the distance or other wireless interferences. Maybe HA is able to detect the lock but when trying to communicate the connection is not stable enough.

I think that's precisely it! Was facing the same issue as @guibar - distance to door about 6-7m with no direct line of sight... after attaching an external bt dongle to my RPI4 it not only found the device but also successfully paired. 

Thank you so much for this awesome integration @ronengr!

ronengr, 2024-04-05:
@DeckardCain2014 
please try the latest version. this issue should be fixed now.

closed 2024-04-05 by ronengr

Mafin-dev, 2024-04-06:
> @DeckardCain2014 please try the latest version. this issue should be fixed now.

No. It is not. And there are still no useful Messages in the log.

Mafin-dev, 2024-04-20:
My full Logs after a fresh HA restart:

[image]

ronengr, 2024-04-20:
@Mafin-dev 
i think you forgot to attach the logs...

for reference, this is how you can get the logs:
https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics

brusarp, 2025-04-14:
Hi @ronengr !
I'm able to reproduce the issue (yay!)...
Since the integration is not configured correctly I can't enable debugging or download diagnostics (or let me know how I can help you).
I also managed to get the Smartlock to work on the first try. After that it seemed that it was not communicating correctly and I tried to reconfigure it. From that moment on nothing:
- HA detects it and suggests that I configure it
- I uninstalled and reinstalled the integration several times
- I tried to factory reset the SmartLock
- every time now a pop-up appears with the word "Error" without any additional information
- I'm using a NUC11 so I tried both using Bluetooth and via Bluetooth Proxy with ESP32

The Nuki SmatLock firmware is the last 3.10.7 (but on first try works correctly with all the information reported)

brusarp, 2025-04-15:
> Hi [@ronengr](https://github.com/ronengr) ! I'm able to reproduce the issue (yay!)... Since the integration is not configured correctly I can't enable debugging or download diagnostics (or let me [...]

Nevermind, solved! I removed the Nuki from the door and put it in front of the NUC to pair it. Now it seems to work again and to be safe I added a BLE Proxy.