ronengr/hass_nuki_bt #21: Can't connect my Nuki lock

diogo-madeira, 2023-10-09:
### System Health details

## System Information

version | core-2023.10.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.5
os_name | Linux
os_version | 6.1.45
arch | x86_64
timezone | America/Recife
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4355
Installed Version | 1.33.0
Stage | running
Available Repositories | 1376
Downloaded Repositories | 47

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 42

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.10.0
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 30.8 GB
disk_used | 11.3 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), Mosquitto broker (6.3.1), Samba share (10.0.2), Home Assistant Google Drive Backup (0.111.1), Duck DNS (1.15.0), Advanced SSH & Web Terminal (15.0.8), AppDaemon (0.13.5), Node-RED (14.6.1), Zigbee2MQTT (1.33.1-1), Zigbee2MQTT Edge (edge), ESPHome (2023.9.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 33
views | 20
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 30 de setembro de 2023 às 17:06
-- | --
current_recorder_run | 9 de outubro de 2023 às 14:04
estimated_db_size | 2773.89 MiB
database_engine | mysql
database_version | 10.11.5

</details>

<details><summary>Sonoff</summary>

version | 3.5.2 (2ad1cd7)
-- | --
cloud_online | 1 / 2
local_online | 1 / 1

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I tried to mannually pair my nuki 2.0 lock, but I get a error without anything writen, only the OK button.
In the HA log I found this:

Logger: homeassistant.helpers.frame
Source: helpers/frame.py:103
First occurred: 14:07:36 (1 occurrences)
Last logged: 14:07:36

Detected integration that attempted to call BleakClient with an address instead of a BLEDevice. Please report issue to the custom integration author for hass_nuki_bt using this method at custom_components/hass_nuki_bt/config_flow.py, line 151: ret = await device.pair()

### Reproduction steps

1. manually pair a nuki lock.
2. put the MAC address
3. click "pair lock automatic (recommended)"

### Debug logs

```text
Logger: homeassistant.helpers.frame
Source: helpers/frame.py:103
First occurred: 14:07:36 (1 occurrences)
Last logged: 14:07:36

Detected integration that attempted to call BleakClient with an address instead of a BLEDevice. Please report issue to the custom integration author for hass_nuki_bt using this method at custom_components/hass_nuki_bt/config_flow.py, line 151: ret = await device.pair()
```

ronengr, 2023-10-17:
Did you try automatic pair?
If HA doesn't automatically recognizes there is a nuki device you can pair, it usually means that HA is not able to connect to that device via BT.

Anyway there is a small bug here in my code that i will fix for next version, but auto pair should still work in current version

diogo-madeira, 2023-10-17:
It doesn't appear anything automatic....
I already use another custom component that use the BT dongle. Is it a problem?

ronengr, 2023-10-17:
if it uses the BT dongle directly than it might be a problem. if it uses HA's BT integration that both should be able to work togther.

diogo-madeira, 2023-10-17:
It's that custom component: https://github.com/custom-components/ble_monitor

diogo-madeira, 2023-10-17:
I found the problem. I need the HA bluetooth integration to work.
Now it appear automatic, but I still have a problem. I get a popup error without any text.
And the log is:

Logger: homeassistant.helpers.frame
Source: helpers/frame.py:103
First occurred: 14:45:52 (1 occurrences)
Last logged: 14:45:52

Detected integration that attempted to call BleakClient with an address instead of a BLEDevice. Please report issue to the custom integration author for hass_nuki_bt using this method at custom_components/hass_nuki_bt/config_flow.py, line 151: ret = await device.pair()

ronengr, 2023-10-17:
The log you are getting is only a warning. From looking at HA code, it should work even if you see this warning.

can you increase log-level and try it again?

ronengr, 2023-11-26:
fixed in 0.0.8

closed 2023-11-26 by ronengr