ronengr/hass_nuki_bt #25: Unable to connect 2 Nuki Smart Lock 3.0 devices

Ayuus, 2023-11-03:
### System Health details

Everything runs as normal

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

I have 2 Nuki devices, both on latest firmware (3.7.7). Each one of them works OK if connected (discovery etc works really well, thnaks!), but the second on cannot be configured. This is independent of which of the 2 devices is connected first.

Is this integration supposed to suppoert multiple Nuki devices?

### Reproduction steps

1. Configure a Nuki Smart Lock 3.0
2. Observe this smart lock works OK
3. Configure another Nuki Smart Lock 3.0
4. Observe that initialisation failed.
...

### Debug logs

```text
In the HA System log:

Detected that custom integration 'hass_nuki_bt' attempted to call BleakClient with an address instead of a BLEDevice at custom_components/hass_nuki_bt/config_flow.py, line 151: ret = await device.pair(), please create a bug report at https://github.com/ronengr/hass_nuki_bt/issues

followed by (5 secs later)

config/custom_components/hass_nuki_bt/config_flow.py:166: RuntimeWarning: coroutine 'NukiDevice.disconnect' was never awaited device.disconnect()
```
```

Ayuus, 2023-11-03:
After configuring a second Bluetooth proxy, both locks work.
Nor\t ideal, as in HA, the bluetooth integration is supposed to be able to connect 5 devices.

closed 2023-11-03 by Ayuus

Ayuus, 2023-11-04:
Hmmm... seems I was wrong, for some reason, both locks now work on the same BT Adapter...

ronengr, 2023-11-27:
Device address is BT address.
note that it should be in the format `38:8B:DC:F4`

Regarding the logs, If you enable debug-logging for the integration before trying to add the 2nd device maybe we will get some more info.

petosiso, 2023-11-27:
Unfortunately, the modified format did not help. The form just flashes and doesn't even throw any error.

My first NUKI device immediately appeared under Homekit integration and works very well. I don't have any Nuki BT integration listed - that seems strange to me. Because of that, I don't even know where to turn on debug-logging.

[image]

I can see only your "New Nuki device" poup form after clicking on "Add integration".

[image]

ronengr, 2023-11-27:
if you go to settings -> devices & services you will see a list of all your enabled integrations.
you can find Nuki there.
depending on you HA version, either click on the 3 dots or on the integration and select "enable debug logging" 
[image]

or click on the integration end then you will see the option inside:
[image]

petosiso, 2023-11-27:
As I mentioned, there is no Nuki integration in list. My first Nuki device is under HomeKit integration.

[image]

Ayuus, 2023-11-27:
> Hi, I have exactly same problem - unable to connect my second Nuki Lock 3.0 device. The second device appears in HA as an iBeacon tracker. Do you have any recommendations?

I ignored iBeacon integration, so maybe that is the difference? Both my Nuki locks work OK via the USB BT adapter I installed on the HA server.

petosiso, 2023-11-27:
No, you didn't understand correctly. I have had HomeKit integration for a long time. I control the gate through it. Today I received 2 Nuki Locks devices, so:

1 - I found your hass_nuki_bt integration and installed it via HACS.
2 - I restarted HA.
3 - HA automatically discovered a new Nuki Lock device throught your integration. I added this device to HA (Configure button)
4 - This first Nuki Lock immediately appeared under HomeKit integration and your integration is not listed.
5 - I cannot add the second Nuki Lock device in any way.

Maybe tomorrow I'll try to remove your integration and repeat the procedure...

Ayuus, 2023-11-27:
> No, you didn't understand correctly. I have had HomeKit integration for a long time. I control the gate through it. Today I received 2 Nuki Locks devices, so:
> 
> 1 - I found your hass_nuki_bt integration and installed it via HACS. 2 - I restarted HA. 3 - HA automatically discovered a new Nuki Lock device throught your integration. To be honest, I am not sure if it was your or homekit integration. I added this device to HA (Configure button) 4 - This first Nuki Lock immediately appeared under HomeKit integration and your integration is not listed. 5 - I cannot add the second Nuki Lock device in any way.
> 
> Maybe tomorrow I'll try to remove your integration and repeat the procedure...
> 
> I feel like the two integrations got mixed together because Nuki natively supports HomeKit - it's kind of like google home, only from Apple. https://nuki.io/en/homekit/

I would ignore Homekit Integration and try again. If necessary, add Nuki Locks manually via Nuki BT Integration.

petosiso, 2023-11-28:
So, I removed the device from HomeKit integration. I deleted your NukiBT integration. Then I reinstalled it and restarted HA. Now I can't even add a single NukiBT device...

Adding NukiBT manually doesn't work either - just like yesterday. After clicking Submit, nothing happens. 
[image]

​I'll probably have to buy Nuki bridge...

petosiso, 2023-11-28:
OK, I found BT address of Nuki lock throught some android app from Google Play. This appears in logs:

File "/usr/src/homeassistant/homeassistant/components/bluetooth/wrappers.py", line 375, in _async_get_best_available_backend_and_device
    raise BleakError(
bleak.exc.BleakError: No backend with an available connection slot that can reach address 54:D2:72:8C:75:EC was found

Ayuus, 2023-11-28:
Sounds like you do not have a Bluetooth adapter installed in your HA server, nor a Bluetooth proxy that can be reached from yoy HA server.