ronengr/hass_nuki_bt #29: Changes of lock state triggered by something else than HA are not updated

guibar, 2023-11-21:
### System Health details

## System Information

version | core-2023.11.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.6
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4973
Installed Version | 1.33.0
Stage | running
Available Repositories | 1404
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 11.1
-- | --
update_channel | stable
supervisor_version | supervisor-2023.11.3
agent_version | 1.6.0
docker_version | 24.0.6
disk_total | 57.8 GB
disk_used | 6.0 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.7.0), Terminal & SSH (9.8.1), SQLite Web (4.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 0
views | 2
mode | storage

[... 280 lines cut ...]
    lock_state = (enum) UNLOCKED 3
    trigger = (enum) BUTTON 2
    current_time = Container: 
        year = 2023
        month = 11
        day = 21
        hour = 18
        minute = 1
        second = 27
    timezone_offset = 0
    critical_battery_state = 148
    config_update_count = 8
    lock_n_go_timer = 0
    last_lock_action = (enum) UNLOCK 1
    last_lock_action_trigger = (enum) BUTTON 2
    last_lock_action_completion_status = (enum) SUCCESS 0
    door_sensor_state = (enum) DOOR_CLOSED 2
    nightmode_active = 1024
    accessory_battery_state = 0
2023-11-21 19:01:13.943 INFO (MainThread) [pyNukiBT.nuki] Nuki: 54:D2:72:96:44:7C, RSSI: -67
2023-11-21 19:05:39.770 ERROR (MainThread) [homeassistant.components.hassio.handler] Client error on /network/info request Cannot connect to host 172.30.32.2:80 ssl:default [No such file or directory]
2023-11-21 19:05:39.771 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /network/info - 
2023-11-21 19:09:51.825 INFO (MainThread) [pyNukiBT.nuki] Nuki: 54:D2:72:96:44:7C, RSSI: -77
2023-11-21 19:09:51.826 INFO (MainThread) [custom_components.hass_nuki_bt] Discovered bluetooth device: {'name': 'Nuki_2E96447C', 'address': '54:D2:72:96:44:7C', 'rssi': -77, 'manufacturer_data': {76: b'\x02\x15\xa9.\xe2\x00U\x01\x11\xe4\x91l\x08\x00 \x0c\x9af.\x96D|\xc8'}, 'service_data': {}, 'service_uuids': [], 'source': 'DC:A6:32:7F:D0:9A', 'advertisement': AdvertisementData(local_name='Nuki_2E96447C', manufacturer_data={76: b'\x02\x15\xa9.\xe2\x00U\x01\x11\xe4\x91l\x08\x00 \x0c\x9af.\x96D|\xc8'}, rssi=-77), 'device': BLEDevice(54:D2:72:96:44:7C, Nuki_2E96447C), 'connectable': True, 'time': 1022.45353807}
```

ronengr, 2023-11-26:
On my device the info is updated even if I open/close the device from the button or the nuki app or any other way.

What BT connection are you using? do you have a BT dongle on the HA PC (what dongle?) or do you use a proxy?

sometime it can take a few seconds for HA to get the update.

I see in the logs that the nuki extension got the state update correctly:

     last_lock_action = (enum) UNLOCK 1
     last_lock_action_trigger = (enum) BUTTON 2
     last_lock_action_completion_status = (enum) SUCCESS 0

but i'm not sure why it is not propagating to HA.

I will look into this a bit more.

guibar, 2023-11-27:
Hi and thanks for looking into this.
I use the native bluetooth of the raspberry pi 4.
I can do more tests to see if the problem is consistent.

Navpoint85, 2024-01-13:
Hello, same problem here. Running standard BT integration with HA Green and BT dongle. Are there any news?

BKingGSXR1000, 2024-01-17:
I had no idea it even SHOULD have those features!
I connected a ASUS USB-BT400 Nano Bluetooth-Stick to my Home Assistant Green on a USB extension cord. I can lock/unlock my Nuki lock with maybe 1-3 seconds delay, which is nice.
The signal strength according to HA changes between -71 and 80 dBM. Don't know if that's good or bad.

But the lock state entity bare ever works, no matter if I press the button on the lock, use the HA-App or do it manually.
As far as I can tell it just runs into a timeout.

`2024-01-17 11:11:49.835 INFO (MainThread) [pyNukiBT.nuki] Nuki: 54:D2:72:B9:52:45, RSSI: -74
2024-01-17 11:11:49.835 INFO (MainThread) [pyNukiBT.nuki] Querying Nuki state
2024-01-17 11:11:54.825 WARNING (MainThread) [homeassistant.components.media_player] Updating aquostv media_player took longer than the scheduled update interval 0:00:10
2024-01-17 11:11:58.265 DEBUG (MainThread) [pyNukiBT.nuki] Services ['a92ee200-5501-11e4-916c-0800200c9a66 (Handle: 12): Unknown', '0000003e-0000-1000-8000-0026bb765291 (Handle: 44): Unknown', '00000044-0000-1000-8000-0026bb765291 (Handle: 151): Unknown', '00000055-0000-1000-8000-0026bb765291 (Handle: 96): Unknown', '00000045-0000-1000-8000-0026bb765291 (Handle: 119): Unknown', '000000a2-0000-1000-8000-0026bb765291 (Handle: 30): Unknown', 'a92ee100-5501-11e4-916c-0800200c9a66 (Handle: 8): Unknown', '0000180a-0000-1000-8000-00805f9b34fb (Handle: 21): Device Information', '00000096-0000-1000-8000-0026bb765291 (Handle: 164): Unknown']
2024-01-17 11:11:58.266 DEBUG (MainThread) [pyNukiBT.nuki] Characteristics ['a92ee203-5501-11e4-916c-0800200c9a66 (Handle: 19): Unknown', 'a92ee202-5501-11e4-916c-0800200c9a66 (Handle: 16): Unknown', 'a92ee201-5501-11e4-916c-0800200c9a66 (Handle: 13): Unknown', '00000014-0000-1000-8000-0026bb765291 (Handle: 47): Hardcopy Data Channel', '00000030-0000-1000-8000-0026bb765291 (Handle: 67): Unknown', '000000a6-0000-1000-8000-0026bb765291 (Handle: 82): Unknown', '00000023-0000-1000-8000-0026bb765291 (Handle: 62): Unknown', '00000053-0000-1000-8000-0026bb765291 (Handle: 77): Unknown', '00000052-0000-1000-8000-0026bb765291 (Handle: 72): Unknown', '00000021-0000-1000-8000-0026bb765291 (Handle: 57): Unknown', '00000020-0000-1000-8000-0026bb765291 (Handle: 52): Unknown', '00000220-0000-1000-8000-0026bb765291 (Handle: 91): Unknown', 'e604e95d-a759-4817-87d3-aa005083a0d1 (Handle: 45): Unknown', '00000019-0000-1000-8000-0026bb765291 (Handle: 154): AVDTP', 'e604e95d-a759-4817-87d3-aa005083a0d1 (Handle: 152): Unknown', '00000037-0000-1000-8000-0026bb765291 (Handle: 159): Unknown', '0000004c-0000-1000-8000-0026bb765291 (Handle: 99): Unknown', '0000004e-0000-1000-8000-0026bb765291 (Handle: 104): Unknown', '00000050-0000-1000-8000-0026bb765291 (Handle: 114): Unknown', '0000004f-0000-1000-8000-0026bb765291 (Handle: 109): Unknown', 'e604e95d-a759-4817-87d3-aa005083a0d1 (Handle: 97): Unknown', '0000001d-0000-1000-8000-0026bb765291 (Handle: 128): Unknown', 'e604e95d-a759-4817-87d3-aa005083a0d1 (Handle: 120): Unknown', '0000001e-0000-1000-8000-0026bb765291 (Handle: 137): MCAP Control Channel', '00000023-0000-1000-8000-0026bb765291 (Handle: 146): Unknown', '000000a5-0000-1000-8000-0026bb765291 (Handle: 122): Unknown', '00000037-0000-1000-8000-0026bb765291 (Handle: 33): Unknown', '000000a5-0000-1000-8000-0026bb765291 (Handle: 38): Unknown', 'e604e95d-a759-4817-87d3-aa005083a0d1 (Handle: 31): Unknown', 'a92ee101-5501-11e4-916c-0800200c9a66 (Handle: 9): Unknown', '00002a26-0000-1000-8000-00805f9b34fb (Handle: 26): Firmware Revision String', '00002a25-0000-1000-8000-00805f9b34fb (Handle: 22): Serial Number String', '00002a24-0000-1000-8000-00805f9b34fb (Handle: 28): Model Number String', '00002a27-0000-1000-8000-00805f9b34fb (Handle: 24): Hardware Revision String', '00000079-0000-1000-8000-0026bb765291 (Handle: 179): Unknown', 'e604e95d-a759-4817-87d3-aa005083a0d1 (Handle: 165): Unknown', '0000008f-0000-1000-8000-0026bb765291 (Handle: 173): Unknown', '00000068-0000-1000-8000-0026bb765291 (Handle: 167): Unknown', '00000023-0000-1000-8000-0026bb765291 (Handle: 185): Unknown']
2024-01-17 11:11:58.321 INFO (MainThread) [pyNukiBT.nuki] Connected
2024-01-17 11:11:58.323 INFO (MainThread) [pyNukiBT.nuki] sending encrypted command REQUEST_DATA
2024-01-17 11:11:58.323 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1
2024-01-17 11:11:58.324 INFO (MainThread) [pyNukiBT.nuki] Connected
2024-01-17 11:11:58.324 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2024-01-17 11:11:58.366 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1
2024-01-17 11:11:58.441 DEBUG (MainThread) [pyNukiBT.nuki] Notification handler: a92ee202-5501-11e4-916c-0800200c9a66 (Handle: 16): Unknown, data: bytearray(b'\xee\r\x1d\x97f\xf9`r\xb46<\xfa\xca\x05\xc1\xc9\xef\xe3;q\xb1\xbc\x05c\xe1\\4\x002\x00\x8ehUY\x18`\xe8K\xdaG\xa5\x1e\xd3d<\x92\xa3b\x0b0\xef\xbc\xb0\xa2\xd21\xed\xf4\x85\x87\n\xf4\x19\xe5\xe5Q\xce\xb7\xbaB\xf2l\xf5J\x94ef\x9br\xe9')
2024-01-17 11:11:58.444 DEBUG (MainThread) [pyNukiBT.nuki] State: Container: 
    nuki_state = (enum) DOOR_MODE 2
    lock_state = (enum) UNLOCKED 3
    trigger = (enum) BUTTON 2
    current_time = Container: 
        year = 2024
        month = 1
        day = 17
        hour = 10
        minute = 11
        second = 59
    timezone_offset = 0
    critical_battery_state = 128
    config_update_count = 10
    lock_n_go_timer = 0
    last_lock_action = (enum) UNLOCK 1
    last_lock_action_trigger = (enum) BUTTON 2
[... cut ...]
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak/__init__.py", line 605, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/bleak/backends/bluezdbus/client.py", line 146, in connect
    async with async_timeout(timeout):
  File "/usr/local/lib/python3.11/asyncio/timeouts.py", line 111, in __aexit__
    raise TimeoutError from exc_val
TimeoutError`

alfredocdmiranda, 2024-02-07:
I am facing similar problem about the update state. If I lock/unlock pressing the button on my Nuki, HA does not get the correct state. I didn't dig into the logs yet.

ronengr, 2024-04-05:
@BKingGSXR1000 
from the logs it seems HA backend has issues connecting to the lock.
probably poor connection quality (to far from the lock, too much interference etc).

If there is no connection or the connection times-out, there is nothing i can do.

ronengr, 2024-04-05:
@guibar @alfredocdmiranda @Navpoint85 
can you please check if there is still an issue in latest version?
there were some fixes in the state_update mechanism that should have solved this.

closed 2024-04-21 by ronengr