ronengr/hass_nuki_bt #39: Nuki 3.0 - Fails to load entity after pairing "AttributeError: 'NukiDevice' object has no attribute '_const'"

SublimePotato, 2024-02-29:
### System Health details

## System Information

version | core-2024.2.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 6.1.73-haos-raspi
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4800
Installed Version | 1.34.0
Stage | running
Available Repositories | 1467
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.02.0
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 457.7 GB
disk_used | 19.1 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.3.0), AdGuard Home (5.0.3), Home Assistant Google Drive Backup (0.112.1), Mosquitto broker (6.4.0), Terminal & SSH (9.9.0), Studio Code Server (5.15.0), Grafana (9.1.3), InfluxDB (5.0.0), Vaultwarden (Bitwarden) (0.21.1), Nginx Proxy Manager (1.0.1), MariaDB (2.6.1), Log Viewer (0.17.0), Zigbee2MQTT (1.35.3-1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 14
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 24. Februar 2024 um 11:47
-- | --
current_recorder_run | 29. Februar 2024 um 11:33
estimated_db_size | 531.07 MiB
database_engine | sqlite
database_version | 3.44.2
[... 255 lines cut ...]
2024-02-29 11:44:36.017 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1
2024-02-29 11:44:36.017 INFO (MainThread) [pyNukiBT.nuki] Connected
2024-02-29 11:44:36.017 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2024-02-29 11:44:36.042 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1
2024-02-29 11:44:36.045 DEBUG (MainThread) [pyNukiBT.nuki] Notification handler: a92ee202-5501-11e4-916c-0800200c9a66 (Handle: 16): Unknown, data: bytearray(b'\t\xcaU\xc3\xa2\xc2\xbeA\xf5\xfb\xe8\xd9\xce{x\xcb\xbc[\x95\x06\xdb\xc3\x91q\xe0\\4\x00M\x00p\xb1e\xd5gL\x1fe\xb8w\xea\xa8\xf7\xbc\x08\x97~#\xb7E[\xeeS\xcc\x97s\x91H\xcf\xc8\xc6@\x88\xfc\xbbfm5&\xfd.2\xc4R\xd5\xf4\xb5\xdc3\xd3~\xbe\xa4_\xdf?\x84\xd9\t\xd4v\xf8\x85\x08F\x03\xb7\xaf\x85\t\t\x07K\xa3\xf7\xd1J')
2024-02-29 11:44:36.073 DEBUG (MainThread) [pyNukiBT.nuki] Notification handler: a92ee202-5501-11e4-916c-0800200c9a66 (Handle: 16): Unknown, data: bytearray(b"\x059\x90\xd0w-=\x1eU\xdb\xdfj\x16\x7f\xa3\x00/!V\\\xc7\x8b*\xbc\xe0\\4\x00\x19\x00\xfb\xec\xff5\xb9\xc9\xeae\xf1j\xad|\xac\xf3\x00\xb0Q\'\xde\xa1\x1d\xe1r\t\xae")
2024-02-29 11:44:36.077 DEBUG (MainThread) [pyNukiBT.nuki] Last action: COMPLETED
2024-02-29 11:44:36.078 DEBUG (MainThread) [pyNukiBT.nuki] Container: 
    status = (enum) COMPLETED 0
2024-02-29 11:44:36.079 DEBUG (MainThread) [pyNukiBT.nuki] [Container(index=13, timestamp=Container(year=2024, month=2, day=29, hour=9, minute=52, second=16), auth_id=4294967295, name=u'', type=uEnumIntegerString.new(4, 'INITIALIZATION_RUN'), data=Container(lock_action=uEnumIntegerString.new(0, 'NONE'), trigger=uEnumIntegerString.new(0, 'SYSTEM'), flags=0, completion_status=uEnumIntegerString.new(254, 'OTHER_ERROR'), padding=None))]
2024-02-29 11:44:36.080 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Nuki_2E02EC7C for hass_nuki_bt
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 444, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 86, in async_setup_entry
    if not await coordinator.async_wait_ready():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 151, in async_wait_ready
    await self._async_update()
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 119, in _async_update
    elif logs[0].index > self.last_nuki_log_entry.index:
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'index'
```

leranp, 2024-03-17:
Same here, i suspect that this is a HA version issue, i started after the update of 2024.3.1

ambernardino, 2024-03-18:
Same here, but I think this is a bluetooth connection issue. Did you disabled bluetooth pairing in the lock itself?

011V32, 2024-03-26:
Same for me. Hope it get an fix 😟

PR #44 (2024-03-30): Fix issue when no last_nuki_log_entry exists
fixes #39
closed 2024-03-30 by ronengr