ronengr/hass_nuki_bt #40: Stuck on initialization - Smart Lock 4.0 Pro

ponchofiesta, 2024-03-05:
### System Health details

## System Information

version | core-2024.2.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.1
os_name | Linux
os_version | 6.6.16-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4936
Installed Version | 1.34.0
Stage | running
Available Repositories | 1400
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.0
-- | --
update_channel | stable
supervisor_version | supervisor-2024.02.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 30.8 GB
disk_used | 5.1 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.9.0), Samba share (12.3.0), OpenThread Border Router (2.5.0), Silicon Labs Multiprotocol (2.4.4), Matter Server (5.4.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 7
views | 3
mode | storage

[... 324 lines cut ...]
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/active_update_coordinator.py", line 129, in _async_poll
    self.data = await self._async_poll_data(self._last_service_info)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/active_update_coordinator.py", line 122, in _async_poll_data
    return await self._poll_method(last_service_info)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 110, in _async_update
    await self.device.update_state()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 433, in update_state
    await self.update_config()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 489, in update_config
    msg = await self._send_encrtypted_command(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 189, in _send_encrtypted_command
    return await self._send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 361, in _send_command
    async with async_timeout.timeout(self.command_response_timeout):
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
TimeoutError
```

ponchofiesta, 2024-03-07:
I also tried another Bluetooth USB dongle (TP-Link EB500). Same problem.

ronengr, 2024-03-30:
From the logs, it seems HA cannot find a connection to the device

> bleak.exc.BleakError: No backend with an available connection slot that can reach address 54:D2:72:87:7E:EC was found

try putting the BT closer to the lock

Unfortunately, there is nothing i can do if i can't establish a BT connection

closed 2024-03-30 by ronengr