ronengr/hass_nuki_bt #41: home assistant the status is not updated if locking/unlocking occurs via the nki app on the smartphone

alikarodi, 2024-03-25:
### System Health details

in home assistant the status is not updated (Nuki Lock State) if locking/unlocking occurs via the nki app on a smartphone
thank

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

when I perform an operation with the smartphone it is not synchronized on the home assistant.
if instead I perform the same operation on the home assistant it is synchronized on the home assistant and on the smartphone

Thank you

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
no debug
```

### Diagnostics dump

no diagnostic dump

ronengr, 2024-03-30:
Did you pair HA as "Bridge" or as "App" (Client Type)?
If it is paired as app it might miss some updates from the lock.

alikarodi, 2024-04-04:
as "App".
Thank

closed 2024-04-04 by alikarodi