ronengr/hass_nuki_bt #55: Sporadic no connection after startup

011V32, 2024-05-17:
### System Health details

## System Information

version | core-2024.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.29-haos
arch | x86_64
timezone | Europe/Vienna
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4911
Installed Version | 1.34.0
Stage | running
Available Repositories | 1396
Downloaded Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.05.1
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 468.7 GB
disk_used | 15.9 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.3.1), File editor (5.8.0), RaspberryMatic CCU (3.75.7.20240420), Advanced SSH & Web Terminal (17.3.0), Mosquitto broker (6.4.0), VNC Viewer (1.6.0), ESPHome (2024.5.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 14
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 9, 2024 at 08:06
-- | --
current_recorder_run | May 17, 2024 at 07:38
estimated_db_size | 272.82 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
used_requests | 0
rooftop_site_count | 1

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

It happens from time to time that no connection can be established after restarting HA. After restarting the integration it works successfully every time.
I use an ESP Home Bluetooth Proxy. Maybe it needs more time to start and the integration should wait until it has started and established a connection?
I activated debug logging but haven't gotten any errors since then. I will submit it later after I get an error and it contains additional information.

### Reproduction steps

Restart HA

### Debug logs

```text
2024-05-17 07:38:29.856 ERROR (MainThread) [custom_components.hass_nuki_bt] 54:D2:72:75:B2:FC: Bluetooth error whilst polling: failed to discover services, device disconnected
2024-05-17 07:38:29.858 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Nuki_3575B2FC for hass_nuki_bt
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 575, in async_setup
result = await component.async_setup_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/hass_nuki_bt/__init__.py", line 86, in async_setup_entry
if not await coordinator.async_wait_ready():
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/hass_nuki_bt/coordinator.py", line 152, in async_wait_ready
await self._async_update()
File "/config/custom_components/hass_nuki_bt/coordinator.py", line 114, in _async_update
logs = await self.device.request_log_entries(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 592, in request_log_entries
self._const.NukiCommand.REQUEST_DATA,
^^^^^^^^^^^
AttributeError: 'NukiDevice' object has no attribute '_const'
```

closed 2024-05-24 by ronengr

isibizi, 2024-10-21:
I have still this issue on my setup

ronengr, 2024-11-11:
@isibizi 
With the same error in the log, or has the error changed in the last version?

The error in the logs you provided should not happen in the latest version

011V32, 2024-11-13:
@ronengr 
This is the error which I still get:

`Logger: homeassistant.config_entries
Source: config_entries.py:635
First occurred: 13:13:42 (1 occurrences)
Last logged: 13:13:42

Error setting up entry Nuki_3575B2FC for hass_nuki_bt
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 214, in connect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/dbus_fast/aio/message_bus.py", line 385, in call
    await future
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 635, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 75, in async_setup_entry
    await device.connect()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 378, in connect
    await self._client.connect()
  File "/usr/local/lib/python3.12/site-packages/habluetooth/wrappers.py", line 311, in connect
    connected = await super().connect(**kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/__init__.py", line 615, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/bleak/backends/bluezdbus/client.py", line 151, in connect
    async with async_timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError`

ronengr, 2024-11-13:
It looks like bleak (the Bluetooth stack used in HA) is getting a timeout when trying to connect.
I've added a catch for TimeoutError in #83 (part of version 0.0.16.1). That should maybe help