ronengr/hass_nuki_bt #57: Failure while polling

011V32, 2024-05-24:
### System Health details

## System Information

version | core-2024.5.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.29-haos
arch | x86_64
timezone | Europe/Vienna
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1396
Downloaded Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.05.1
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 468.7 GB
disk_used | 16.1 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.3.1), File editor (5.8.0), RaspberryMatic CCU (3.75.7.20240420), Advanced SSH & Web Terminal (17.3.0), Mosquitto broker (6.4.0), VNC Viewer (1.6.0), ESPHome (2024.5.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 14
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 14, 2024 at 08:52
-- | --
current_recorder_run | May 21, 2024 at 21:56
estimated_db_size | 287.28 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
used_requests | 6
rooftop_site_count | 1

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Error in the log. See Debug log.

### Reproduction steps

I do nothing specific or I couldn't tell yet. It came from time to time...

### Debug logs

```text
This error originated from a custom integration.

Logger: custom_components.hass_nuki_bt
Source: components/bluetooth/active_update_coordinator.py:131
integration: Nuki Lock (documentation, issues)
First occurred: May 23, 2024 at 05:52:17 (1 occurrences)
Last logged: May 23, 2024 at 05:52:17

54:D2:72:75:B2:FC: Failure while polling
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 366, in _send_command
    msg = await self._notify_future
          ^^^^^^^^^^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/active_update_coordinator.py", line 131, in _async_poll
    self.data = await self._async_poll_data(self._last_service_info)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/bluetooth/active_update_coordinator.py", line 124, in _async_poll_data
    return await self._poll_method(last_service_info)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 110, in _async_update
    await self.device.update_state()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 428, in update_state
    msg = await self._send_encrtypted_command(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 193, in _send_encrtypted_command
    return await self._send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 365, in _send_command
    async with async_timeout.timeout(self.command_response_timeout):
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
TimeoutError
```

ronengr, 2024-05-24:
From the description and the error nature (timeout when trying to send command) it seems the issue is with the BT connection.
Could be some wireless interference, weak signal, Nuki is too far away from BT or something like that.

011V32, 2024-05-27:
I trie next week a longer cable for the BT Proxy so I get closer to the lock.
I then report back. Thx so far.

011V32, 2024-06-04:
With the version 0.0.14 I have no issues so far. 👍 
Thx.

closed 2024-06-04 by 011V32