ronengr/hass_nuki_bt #58: Error executing service: ServiceCall lock.lock

011V32, 2024-05-24:
### System Health details

## System Information

version | core-2024.5.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.29-haos
arch | x86_64
timezone | Europe/Vienna
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1396
Downloaded Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.3
-- | --
update_channel | stable
supervisor_version | supervisor-2024.05.1
agent_version | 1.6.0
docker_version | 25.0.5
disk_total | 468.7 GB
disk_used | 16.1 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (12.3.1), File editor (5.8.0), RaspberryMatic CCU (3.75.7.20240420), Advanced SSH & Web Terminal (17.3.0), Mosquitto broker (6.4.0), VNC Viewer (1.6.0), ESPHome (2024.5.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 14
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 14, 2024 at 08:52
-- | --
current_recorder_run | May 21, 2024 at 21:56
estimated_db_size | 287.28 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
used_requests | 6
rooftop_site_count | 1

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Error in the log. See Debug log.

### Reproduction steps

I use HomeKit and have integrated this lock. At 7:30 it automatically locks the door if it is not locked.

### Debug logs

```text
Logger: homeassistant.core
Source: core.py:2758
First occurred: May 23, 2024 at 07:30:20 (1 occurrences)
Last logged: May 23, 2024 at 07:30:20

Error executing service: <ServiceCall lock.lock (c:01HYHYTYF7S23TG94TMVSPBE2M): entity_id=['lock.nuki_3575b2fc_lock']>
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 366, in _send_command
    msg = await self._notify_future
          ^^^^^^^^^^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 2758, in _run_service_call_catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 2779, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 975, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1047, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/lock/__init__.py", line 213, in async_handle_lock_service
    await self.async_lock(**self.add_default_code(kwargs))
  File "/config/custom_components/hass_nuki_bt/lock.py", line 49, in async_lock
    await self.async_lock_action(NukiLockConst.LockAction.LOCK)
  File "/config/custom_components/hass_nuki_bt/entity.py", line 76, in async_lock_action
    msg = await self.device.lock_action(action, name_suffix=user_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 464, in lock_action
    msg = await self._send_encrtypted_command(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 193, in _send_encrtypted_command
    return await self._send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 365, in _send_command
    async with async_timeout.timeout(self.command_response_timeout):
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 141, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.12/site-packages/async_timeout/__init__.py", line 228, in _do_exit
    raise asyncio.TimeoutError
TimeoutError
```

ronengr, 2024-05-24:
It seems there is an issue sending the command at the BT level.
Is the Nuki device close enough to the BT dongle? are there any wireless interferences?

011V32, 2024-05-24:
I use an Olimex PoE ESP32 as Bluetooth Proxy which is approx. 1.5m away from the lock.

ronengr, 2024-05-25:
how is the BT signal strength?
maybe there are some wireless interferences? 

You can try to increase the timeout in hass_nuki_bt (unfortunately not user configurable at the moment, so  you will have to change the code)
or you can try to change the energy-saving mode in the nuki lock, and increase the BT scan interval of the lock.

another option is to add some retry mechanism, but i'm not sure if it should be in the integration or in the automation.
what do you think?

011V32, 2024-05-27:
The signal strength varies from -45 to -105:
[image]
I trie next week a longer cable for the BT Proxy so I get closer to the lock, but there are no real disturbanced in the near...
I then report back. Thx so far.

ronengr, 2024-05-31:
@011V32 
I think that -105 is a bit low, and can cause sporadic issues. I've noticed that when i have -90 and below I get a lot more connection issues than when it is above -70

Anyway, I've just released a new version that improves the retry mechanism, so even if you have sporadic connection issues with the new version things should work a lot better.

Please let me know if the new version fixed your issue.

011V32, 2024-05-31:
The issue is also with good condition. Atm i'm on holiday. I'll check it next week.
Thx

011V32, 2024-06-04:
With the version 0.0.14 I have no issues so far. 👍 
Thx.

closed 2024-06-04 by 011V32