ronengr/hass_nuki_bt #63: Cannot load component V0.0.14 and V0.0.13 with Nuki V4

cicero200272, 2024-06-03:
### System Health details

## System Information

version | core-2024.5.2
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 5.10.103-v7l+
arch | armv7l
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4421
Installed Version | 1.34.0
Stage | running
Available Repositories | 1388
Downloaded Repositories | 28

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 21. September 2024 um 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
remote_server | eu-central-1-7.ui.nabu.casa
certificate_status | ready
instance_id | 35c5ebd370f5454f90f48a8a563ac01f
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 14
views | 7
mode | yaml

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 15. Mai 2024 um 10:44
-- | --
current_recorder_run | 3. Juni 2024 um 11:21
estimated_db_size | 200.08 MiB
database_engine | mysql
database_version | 10.6.11

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Nuki Device type

Nuki V4 non pro (no bridge)

### hass_nuki_bt version

V0.0.14 and V0.0.13

### Describe the issue

V0.0.14 and V0.0.13 do not load when HA starts up. V0.0.12 works without errors.

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
2024-06-03 11:19:10.026 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'hass_nuki_bt': Unable to import component: cannot import name 'OffsettedEnd' from 'construct' (/usr/local/lib/python3.12/site-packages/construct/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 981, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1041, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 144, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 16, in <module>
    from pyNukiBT import NukiDevice, NukiConst
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/__init__.py", line 1, in <module>
    from .const import NukiConst, NukiLockConst, NukiOpenerConst, NukiErrorException
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/const.py", line 2, in <module>
    from construct import Bit, BitStruct, Optional, Padding, OffsettedEnd, Struct, Byte, Enum, Int8ul, Int16ul, Int32ul, Int8sl, Int16sl, Float32l, PaddedString, Bytes, Switch, GreedyBytes, this
ImportError: cannot import name 'OffsettedEnd' from 'construct' (/usr/local/lib/python3.12/site-packages/construct/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 322, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1001, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 993, in async_get_component
    comp = self._get_component()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1041, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 144, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 16, in <module>
    from pyNukiBT import NukiDevice, NukiConst
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/__init__.py", line 1, in <module>
    from .const import NukiConst, NukiLockConst, NukiOpenerConst, NukiErrorException
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/const.py", line 2, in <module>
    from construct import Bit, BitStruct, Optional, Padding, OffsettedEnd, Struct, Byte, Enum, Int8ul, Int16ul, Int32ul, Int8sl, Int16sl, Float32l, PaddedString, Bytes, Switch, GreedyBytes, this
ImportError: cannot import name 'OffsettedEnd' from 'construct' (/usr/local/lib/python3.12/site-packages/construct/__init__.py)
```

ronengr, 2024-06-03:
Interesting.
It seems you have and old version of "construct" and for some reason HA didn't automatically install construct 2.10.70, even though it is in the dependencies of pyNukiBT

Did you get any errors during installation?
Do you have any other integration that uses the construct package, and maybe force-set it to a lower version?

As a workaround, you can try manually upgrading the construct package to 2.10.70. But it is interesting to find out why this was not done automatically by HA

cicero200272, 2024-06-04:
I manually upgraded construct to 2.10.70 (was 2.10.68), but I still get the same error during startup. No other components are using this package afaik...

stones80, 2024-06-07:
I have the same problem. Thanks for the support

yyogo, 2024-06-12:
Looks like this is due to a conflict with another HA component:

https://github.com/home-assistant/core/blob/master/requirements_all.txt#L672-L673

closed 2024-06-12 by cicero200272

scr4t3r, 2024-07-07:
unfortunately I have the same issue. not sure if it helps, but I have the standard xiaomi integration active (for roborock). @cicero200272 can you post here how you where able to change the xiaomi dependency from 2.10.68 to 2.10.70?

scr4t3r, 2024-08-19:
> @scr4t3r @cryptedx you can try using my fork in the meantime https://github.com/yyogo/hass_nuki_bt

thanks @yyogo your fork works! would be amazing if it were implemented in the main version
@cicero200272 would you mind re-opening the issue?

ronengr, 2024-11-11:
@scr4t3r 
fixed in latest version (0.0.15)

ardstuebck, 2024-11-15:
Hi, I updated to 16.2 but still get the error:
```

Logger: pyNukiBT.const
Quelle: custom_components/hass_nuki_bt/__init__.py:17

pyNukiBT requires constructs version 2.10.70 or above, but version 2.10.68 is installed
pyNukiBT will try to work with the old version. Some issues might occur.

```

```

Logger: aiohttp.server
Quelle: /usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py:448

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 477, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 559, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 122, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 367, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 410, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 517, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/config_flow.py", line 158, in async_step_pair
    await device.connect()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 381, in connect
    await self._client.connect()
  File "/usr/local/lib/python3.12/site-packages/habluetooth/wrappers.py", line 286, in connect
    wrapped_backend = self._async_get_best_available_backend_and_device(manager)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/habluetooth/wrappers.py", line 395, in _async_get_best_available_backend_and_device
    raise BleakError(
bleak.exc.BleakError: No backend with an available connection slot that can reach address 54:D2:72:A0:CC:7C was found

```

ronengr, 2024-11-15:
> Hi, I updated to 16.2 but still get the error:
> 
> ```
> 
>   File "/usr/local/lib/python3.12/site-packages/habluetooth/wrappers.py", line 395, in _async_get_best_available_backend_and_device
>     raise BleakError(
> bleak.exc.BleakError: No backend with an available connection slot that can reach address 54:D2:72:A0:CC:7C was found
> ```
This is a HomeAssistant Bluetooth error. It means HA has (at the moment at least) no way to communicate with the device.
Unfortunately, there is nothing hass_nuki_bt can do about it. If the underling BT stack cannot create a connection the integration cannot work.
You can try getting your BT device closer to the lock, or other methods to try and improve BT connectivity .

scr4t3r, 2024-11-16:
@ronengr unfortunately i am in the same boat as @ardstuebck :(

scr4t3r, 2024-11-17:
> @scr4t3r Are you getting the same BleakError that it is unable to reach your Nuki device?
> 
> Are you using a BT proxy or an onboard BT on the HA machine? How far is the BT from the lock?

it looks different to me. i m getting this one. it's been working with the version by @yyogo and i ve "sidegrated" to it again and it still works

`2024-11-16 16:37:35.347 WARNING (ImportExecutor_0) [pyNukiBT.const] pyNukiBT requires constructs version 2.10.70 or above, but version 2.10.68 is installed
2024-11-16 16:37:35.348 WARNING (ImportExecutor_0) [pyNukiBT.const] pyNukiBT will try to work with the old version. Some issues might occur.
2024-11-16 16:37:37.020 ERROR (MainThread) [pyNukiBT.nuki] Checksum error in incoming message Error in path (parsing) -> crc
wrong checksum, read 21248, computed 12324
2024-11-16 16:37:37.021 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Nuki_3D7041A5 for hass_nuki_bt
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 92, in async_setup_entry
    if not await coordinator.async_wait_ready():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 137, in async_wait_ready
    await self._async_update()
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 115, in _async_update
    await self.device.update_state()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 419, in update_state
    msg = await self._send_encrtypted_command(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 173, in _send_encrtypted_command
    return await self._send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 348, in _send_command
    msg = await self._notify_future
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 240, in _notification_handler
    msg = self._parse_message(bytes(data), encrypted)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 185, in _parse_message
    msg = self._const.NukiMessage.parse(self._decrypt_message(data))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 288, in parse
    def __init__(self):
               ^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 300, in parse_stream
    return {k.replace("_","-").rstrip("-"):v for k,v in d.items()}
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 312, in _parsereport
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 2120, in _parse
    >>> x = object
                   
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 312, in _parsereport
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 2653, in _parse
    b'\x00\x01\x02\x03\x04\x05\x06\x07\x08'
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 312, in _parsereport
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/const.py", line 1020, in _parse
    raise construct.core.ChecksumError(
construct.core.ChecksumError: Error in path (parsing) -> crc
wrong checksum, read 21248, computed 12324
`

ronengr, 2024-11-17:
@scr4t3r 
this is a completely different issue.
It is caused by a combination of 2 problems:
1. You have constructs version 2.10.68 installed, while pyNukiBT requires 2.10.70.
  this is due to a dependency issue in HA. There is another extension that, for some reason, requires specifically 2.10.68 (instead a more relax requirement, as it will work perfectly ok with 2.10.70). Unfortunately the only way to fix is is to fix the extension requirements in HA, which is something I cannot do.
2. You are using a Nuki lock version that i don't have (I guess Nuki4?). In this version they added some fields in the data transmitted that I am not aware of.
  Fortunately, this is something I can fix. If you will help me.

I'm guessing that when you use @yyogo 's version, you see log lines similar to:

> 2024-11-17 12:00:00.000 WARNING (MainThread) [pyNukiBT.nuki] Got unexpected message length for command KEYTURNER_STATES. Got 1 unknown bytes with value: b'\x00'
2024-11-17 12:00:00.000 WARNING (MainThread) [pyNukiBT.nuki] Got unexpected message length for command CONFIG. Got 1 unknown bytes with value: b'\x00'
2024-11-17 12:00:00.000 WARNING (MainThread) [pyNukiBT.nuki] Got unexpected message length for command LOG_ENTRY. Got 3 unknown bytes with value: b'\x00\x00\x00'

For different commands (with different number of bytes and different values).

If you could open a new issue, and add all the logs you get for "unexpected message length for command", I could fix the expected message structure, so pyNukiBT will work with your lock even with the old version of constructs

yyogo, 2024-11-18:
@scr4t3r try removing the package entirely, delete the directory from the `/config/custom_components/hass_nuki_bt` and then reinstall. there may be some leftovers from my version that are interfering with the mainline version

ronengr, 2024-11-18:
@yyogo 
from the logs it seems that construct version 2.10.68 is installed. This version doesn't support OffsettedEnd (it was introduced in version 2.10.70).
With OffsettedEnd pyNukiBT is able to parse messages even if it has some fields added that it is not aware of. All the access data is simply put in the "unknown" field of the message struct (and a warning is printed).
Without OffsettedEnd pyNukiBT needs to know the full structure of the message, and if there are some new fields at the end it will fail.
This can easily be fixed by adding the missing fields to the message struct in pyNukiBT, but in order to do that i need to know what fields are missing, and that data can be obtained from the warning log i mentioned before.

The problem is that these extra fields seem to exist only on Nuki4 which i don't have, so I cannot test by myself. With Nuki3 there are no such unknown fields.

yyogo, 2024-11-18:
Ah, I see. Maybe the dependency could be updated upstream in home-assistant/core? 

Regardless I happen to have a v4 lock (BT only, no opener, no bridge), so I'll try to look into the missing fields and open a PR.

ronengr, 2024-11-18:
> Maybe the dependency could be updated upstream in home-assistant/core?
Maybe. I have no idea how they handle dependency versions there.

> Regardless I happen to have a v4 lock (BT only, no opener, no bridge), so I'll try to look into the missing fields and open a PR.

Thanks!
I've started adding some missing fields from what i found in logs sent to me, but i'm not sure if i got all of them.
you can take this as a base:

`diff --git a/pyNukiBT/const.py b/pyNukiBT/const.py
index ac93aea..265e013 100644
--- a/pyNukiBT/const.py
+++ b/pyNukiBT/const.py
@@ -481,7 +481,9 @@ class NukiConst:
             "trigger" / self.ActionTrigger,
             "flags" / Int8ul,
             "completion_status" / self.LockActionCompletionStatus,
-            "padding" / Optional(Padding(1)), #Nuki3 has padding, Nuki4 doesn't
+            "padding" / Optional(Padding(1)), #Nuki3 has 1 padding, others may not.
+            Optional(Padding(3)), #Nuki4 padding
+
         )
 
     @functools.cached_property
@@ -739,6 +741,7 @@ class NukiLockConst(NukiConst):
         "nightmode_active" / Int16ul,
         "accessory_battery_state" / Optional(Int8ul),
         Optional(Padding(4)), #this doesn't exist in the documentation, but we see it in real world communications
+        Optional(Padding(1)), #Nuki4 has one more.
     )
 
     Config = Struct(
@@ -925,6 +928,7 @@ class NukiOpenerConst(NukiConst):
         "undocumented" / Int8ul,
         "undocumented2" / Int8ul,
         "has_keypad_v2" / Int8ul,
+        Optional(Padding(1)), #Nuki4 padding
     )
 
     NewConfig = Struct(
`

scr4t3r, 2024-12-02:
> @scr4t3r this is a completely different issue. It is caused by a combination of 2 problems:
>
> 1. You have constructs version 2.10.68 installed, while pyNukiBT requires 2.10.70.
>    this is due to a [...]

hi @ronengr i just went through the logs of the last 3 weeks. i only get the CONFIG (once) and KEYTURNER_STATES (tons of times) errors in exactly the format you write above. none other!

ronengr, 2025-01-08:
@scr4t3r thanks for the info.
I will update pyNukiBT accordingly.

ronengr, 2025-01-08:
@yyogo 
I found a new way to handle older construct versions, which should enable using everything i need for pyNukiBT even with older construct.
things should work much better in hass_nuki_bt version 0.0.17