ronengr/hass_nuki_bt #68: after update    Unable to import component: cannot import name 'OffsettedEnd'

alikarodi, 2024-07-19:
### System Health details

 Core 2024.7.2
Supervisor 2024.06.2
Operating System 12.4
Frontend 20240710.0

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Nuki Device type

SMARTLOCK_1_2

### hass_nuki_bt version

v.0.0.14

### Describe the issue

smart lock not working

### Reproduction steps

not load

### Debug logs

```text
Registratore: homeassistant.config_entries
Fonte: config_entries.py:2832
Prima occorrenza: 18 luglio 2024 alle ore 18:14:18 (1 occorrenze)
Ultima registrazione: 18 luglio 2024 alle ore 18:14:18

Error occurred loading flow for integration hass_nuki_bt: cannot import name 'OffsettedEnd' from 'construct' (/usr/local/lib/python3.12/site-packages/construct/__init__.py)

Registratore: homeassistant.setup
Fonte: setup.py:322
Prima occorrenza: 18 luglio 2024 alle ore 18:14:01 (1 occorrenze)
Ultima registrazione: 18 luglio 2024 alle ore 18:14:01

Setup failed for custom integration 'hass_nuki_bt': Unable to import component: cannot import name 'OffsettedEnd' from 'construct' (/usr/local/lib/python3.12/site-packages/construct/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1007, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1067, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 16, in <module>
    from pyNukiBT import NukiDevice, NukiConst
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/__init__.py", line 1, in <module>
    from .const import NukiConst, NukiLockConst, NukiOpenerConst, NukiErrorException
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/const.py", line 2, in <module>
    from construct import Bit, BitStruct, Optional, Padding, OffsettedEnd, Struct, Byte, Enum, Int8ul, Int16ul, Int32ul, Int8sl, Int16sl, Float32l, PaddedString, Bytes, Switch, GreedyBytes, this
ImportError: cannot import name 'OffsettedEnd' from 'construct' (/usr/local/lib/python3.12/site-packages/construct/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 322, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    self._component_future.result()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1019, in async_get_component
    comp = self._get_component()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1067, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 16, in <module>
    from pyNukiBT import NukiDevice, NukiConst
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/__init__.py", line 1, in <module>
    from .const import NukiConst, NukiLockConst, NukiOpenerConst, NukiErrorException
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/const.py", line 2, in <module>
    from construct import Bit, BitStruct, Optional, Padding, OffsettedEnd, Struct, Byte, Enum, Int8ul, Int16ul, Int32ul, Int8sl, Int16sl, Float32l, PaddedString, Bytes, Switch, GreedyBytes, this
ImportError: cannot import name 'OffsettedEnd' from 'construct' (/usr/local/lib/python3.12/site-packages/construct/__init__.py)
```

yyogo, 2024-07-24:
see also #63

ronengr, 2024-11-11:
@alikarodi 
fixed in latest version (0.0.15)

closed 2024-11-11 by ronengr