ronengr/hass_nuki_bt #70: Can't setup Nuki 1.0

DeamBeam, 2024-07-31:
### System Health details

## System Information

version | core-2024.7.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.6.33-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4868
Installed Version | 1.34.0
Stage | running
Available Repositories | 1388
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.4
-- | --
update_channel | stable
supervisor_version | supervisor-2024.06.2
agent_version | 1.6.0
docker_version | 26.1.4
disk_total | 30.8 GB
disk_used | 8.4 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Vaultwarden (Bitwarden) (0.22.0), Mosquitto broker (6.4.1), Terminal & SSH (9.14.0), Meross Local Broker Addon (0.0.1-alpha49), Zigbee2MQTT (1.39.0-1)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
[... 167 lines cut ...]
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 155, in async_wait_ready
    await self._async_update()
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 112, in _async_update
    await self.device.update_state()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 421, in update_state
    msg = await self._send_encrtypted_command(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 184, in _send_encrtypted_command
    unencrypted = self._const.NukiMessage.build(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 452, in build
    self.build_stream(obj, stream, **contextkw)
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 464, in build_stream
    self._build(obj, stream, context, "(building)")
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 2260, in _build
    buildret = sc._build(subobj, stream, context, path)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 2774, in _build
    return self.subcon._build(obj, stream, context, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 4559, in _build
    return self.subcon._build(obj, stream, context, path)
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'tmpclass' object has no attribute '_build'
```

SlKlElT, 2024-08-03:
Same problem here with nuki 2.0

`Error setting up entry Garagentür for hass_nuki_bt
Error setting up entry Garagentür for hass_nuki_bt
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 586, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 90, in async_setup_entry
    if not await coordinator.async_wait_ready():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 155, in async_wait_ready
    await self._async_update()
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 112, in _async_update
    await self.device.update_state()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 421, in update_state
    msg = await self._send_encrtypted_command(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 184, in _send_encrtypted_command
    unencrypted = self._const.NukiMessage.build(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 452, in build
    self.build_stream(obj, stream, **contextkw)
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 464, in build_stream
    self._build(obj, stream, context, "(building)")
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 2260, in _build
    buildret = sc._build(subobj, stream, context, path)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 2774, in _build
    return self.subcon._build(obj, stream, context, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 4559, in _build
    return self.subcon._build(obj, stream, context, path)
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'tmpclass' object has no attribute '_build'`

SlKlElT, 2024-08-20:
Is this repo also dead?

ronengr, 2024-11-11:
@SlKlElT no, not dead.
I was a bit busy during the last few months, but now I'm back :-)

ronengr, 2024-11-11:
As I don't have Nuki1 or Nuki2 device to test, this kind of issues are to be expected.
It seems the Nuki1 doesn't have accessory_battery_state as part of it's KeyturnerStates message. I will try to fix this for next version.

@SlKlElT 
Could you please add some more logs? I'm guessing there should also be some StreamError in the Nuki2 case.

ronengr, 2024-11-11:
@DeamBeam 
Should be fixed in latest version (0.0.15)
If you get a new error (which might happen, as i don't have a Nuki1 to test), please let me know.

closed 2024-11-11 by ronengr

SlKlElT, 2024-11-11:
> As I don't have Nuki1 or Nuki2 device to test, this kind of issues are to be expected. It seems the Nuki1 doesn't have accessory_battery_state as part of it's KeyturnerStates message. I will try to [...]

i moved to the deprecated addon meanwhile, i will check the new Version and get back to you.

SlKlElT, 2024-11-23:
> As I don't have Nuki1 or Nuki2 device to test, this kind of issues are to be expected. It seems the Nuki1 doesn't have accessory_battery_state as part of it's KeyturnerStates message. I will try to [...]

@ronengr i've tried it now with the Version 0.0.16.2 - still not working

```
Logger: homeassistant.config_entries
Quelle: config_entries.py:635
Erstmals aufgetreten: 16:12:16 (3 Vorkommnisse)
Zuletzt protokolliert: 16:13:33

Error setting up entry Nuki_24BC6684 for hass_nuki_bt
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 635, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 92, in async_setup_entry
    if not await coordinator.async_wait_ready():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 137, in async_wait_ready
    await self._async_update()
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 115, in _async_update
    await self.device.update_state()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 419, in update_state
    msg = await self._send_encrtypted_command(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 173, in _send_encrtypted_command
    return await self._send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 348, in _send_command
    msg = await self._notify_future
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 240, in _notification_handler
    msg = self._parse_message(bytes(data), encrypted)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 185, in _parse_message
    msg = self._const.NukiMessage.parse(self._decrypt_message(data))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 288, in parse
    return self.parse_stream(io.BytesIO(data), **contextkw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 300, in parse_stream
    return self._parsereport(stream, context, "(parsing)")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 2120, in _parse
    subobj = sc._parsereport(stream, context, path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 2653, in _parse
    return self.subcon._parsereport(stream, context, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/const.py", line 1016, in _parse
    hash1 = self.checksumfield._parsereport(stream, context, path)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 1041, in _parse
    data = stream_read(stream, self.length, path)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 91, in stream_read
    raise StreamError("stream read less than specified amount, expected %d, found %d" % (length, len(data)), path=path)
construct.core.StreamError: Error in path (parsing) -> crc
stream read less than specified amount, expected 2, found 1
```

ronengr, 2024-11-24:
@SlKlElT 
Thanks for the feedback, I will lock into this.

ronengr, 2025-01-08:
@SlKlElT
I've just released a new version (0.0.17) that should fix this error.