ronengr/hass_nuki_bt #80: AttributeError: module 'construct.core' has no attribute 'BytesIOWithOffsets'

CrazYoshi, 2024-11-13:
### System Health details

## System Information

version | core-2024.11.1
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.5.0-0.deb12.4-amd64
arch | x86_64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4974
Installed Version | 2.0.1
Stage | running
Available Repositories | 1472
Downloaded Repositories | 16

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 31

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 9
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 8 novembre 2024 alle ore 10:01
-- | --
current_recorder_run | 13 novembre 2024 alle ore 11:19
estimated_db_size | 300.07 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Nuki Device type

Nuki Smart Lock 3.0

### hass_nuki_bt version

0.0.15.1

### Describe the issue

I've updated hass_nuki_bt using HACS from version 0.0.12 that was working correctly to the latest one 0.0.15.1
Now the integration doesn't work anymore and it is not able to initialize properly.

### Reproduction steps

1. Update hass_nuki_bt to latest version from 0.0.12
2. Restart HA

### Debug logs

```text
Registratore: homeassistant.config_entries
Fonte: config_entries.py:635
Prima occorrenza: 11:20:55 (1 occorrenze)
Ultima registrazione: 11:20:55

Error setting up entry Nuki_366EA4FC for hass_nuki_bt
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 635, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/__init__.py", line 90, in async_setup_entry
    if not await coordinator.async_wait_ready():
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 137, in async_wait_ready
    await self._async_update()
  File "/config/custom_components/hass_nuki_bt/coordinator.py", line 115, in _async_update
    await self.device.update_state()
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 419, in update_state
    msg = await self._send_encrtypted_command(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 173, in _send_encrtypted_command
    return await self._send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 348, in _send_command
    msg = await self._notify_future
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 240, in _notification_handler
    msg = self._parse_message(bytes(data), encrypted)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/nuki.py", line 185, in _parse_message
    msg = self._const.NukiMessage.parse(self._decrypt_message(data))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 288, in parse
    def __init__(self):
               ^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 300, in parse_stream
    return {k.replace("_","-").rstrip("-"):v for k,v in d.items()}
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 312, in _parsereport
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 2120, in _parse
    >>> x = object
                   
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 312, in _parsereport
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 2653, in _parse
    b'\x00\x01\x02\x03\x04\x05\x06\x07\x08'
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/construct/core.py", line 312, in _parsereport
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/const.py", line 1036, in _parse_other
    ret = super()._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pyNukiBT/const.py", line 25, in _parse
    substream = construct.core.BytesIOWithOffsets.from_reading(stream, length, path)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'construct.core' has no attribute 'BytesIOWithOffsets'
```

closed 2024-11-13 by ronengr

ronengr, 2024-11-13:
@CrazYoshi 
This should be fixed in latest version (0.0.16)
Could you please check if it is working for you and let me know?

CrazYoshi, 2024-11-13:
Yes, I will. Thanks for your quick response

ronengr, 2024-11-13:
Thanks. I missed this error.
Should now be fixed in 0.0.16.1

CrazYoshi, 2024-11-13:
It's working now! Thanks for your help! 👍