ronengr/hass_nuki_bt #87: Error setting up entry Nuki_xxxxxxxx for hass_nuki_bt

peperoni60, 2025-01-04:
### System Health details

## System Information

version | core-2025.1.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.1
Stage | running
Available Repositories | 1494
Downloaded Repositories | 21

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 30.8 GB
disk_used | 23.3 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | kvm
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.18.0), TasmoAdmin (0.31.2), Terminal & SSH (9.16.0), Cloudflared (5.2.5), Samba Backup (5.2.0), RaspberryMatic CCU (3.79.6.20241122), Mosquitto broker (6.4.1), Whisper (2.4.0), Piper (1.5.2), openWakeWord (1.10.0), Zigbee2MQTT (2.0.0-1)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
[... 132 lines cut ...]
  File "/usr/local/lib/python3.13/site-packages/construct/core.py", line 312, in _parsereport
  File "/usr/local/lib/python3.13/site-packages/construct/core.py", line 2120, in _parse
    >>> x = object
    
  File "/usr/local/lib/python3.13/site-packages/construct/core.py", line 312, in _parsereport
  File "/usr/local/lib/python3.13/site-packages/construct/core.py", line 2653, in _parse
    b'\x00\x01\x02\x03\x04\x05\x06\x07\x08'
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/construct/core.py", line 312, in _parsereport
  File "/usr/local/lib/python3.13/site-packages/pyNukiBT/const.py", line 1016, in _parse
    hash1 = self.checksumfield._parsereport(stream, context, path)
  File "/usr/local/lib/python3.13/site-packages/construct/core.py", line 312, in _parsereport
  File "/usr/local/lib/python3.13/site-packages/construct/core.py", line 1041, in _parse
    >>> d = Bitwise(Struct(
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/construct/core.py", line 91, in stream_read
    Only one parsing class can raise this exception: Error. It can mean the parsing class was merely parsed or built with.
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
construct.core.StreamError: Error in path (parsing) -> crc
stream read less than specified amount, expected 2, found 1
```

### Diagnostics dump

not available

ronengr, 2025-01-08:
thanks for the feedback. 
There was an issue with Nuki2 in the old version, I've just released a new version (0.0.17) that should hopefully fix this error.

peperoni60, 2025-01-10:
Thank you! You did a great job and saved me from spending money for a cloud-bound (🙁) bridge! I use it with a [bluetooth 5.4 dongle with external antenna](https://www.amazon.de/dp/B0D3L662WN) and it works over a distance of more than 10 meters!

ronengr, 2025-01-10:
I'm glad it works for you :-)

closed 2025-01-10 by ronengr