ronengr/hass_nuki_bt #90: Error trying to add Nuki Smart Lock

polcape, 2025-01-11:
### System Health details

## System Information

version | core-2025.1.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4994
Installed Version | 2.0.2
Stage | running
Available Repositories | 1500
Downloaded Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 234.0 GB
disk_used | 23.6 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | MariaDB (2.7.2), Advanced SSH & Web Terminal (20.0.0), InfluxDB (5.0.1), Grafana (10.2.2), Nginx Proxy Manager (1.0.1), Node-RED (19.0.0), phpMyAdmin (0.11.0), Network UPS Tools (0.14.0), Samba share (12.3.2), Studio Code Server (5.18.0), Mosquitto broker (6.4.1), ESPHome Device Builder (2024.12.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 1
views | 6
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 31 dicembre 2024 alle ore 12:26
-- | --
current_recorder_run | 11 gennaio 2025 alle ore 16:27
estimated_db_size | 814.61 MiB
database_engine | mysql
database_version | 10.11.6

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of currently [previous issues](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Nuki Device type

Nuki Smart Lock

### hass_nuki_bt version

0.0.17

### Describe the issue

I get error trying to add the device

### Reproduction steps

1. Device is discovered
2. Click on Add
3. Insert the security PIN setted in Nuki app
4. Choose Bridge mode and click on Pair automatically

### Debug logs

```text
Registratore: aiohttp.server
Fonte: /usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py:451
Prima occorrenza: 16:31:22 (1 occorrenze)
Ultimo accesso: 16:31:22

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 210, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 122, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 341, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 384, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, user_input["next_step_id"], None
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 491, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/config_flow.py", line 158, in async_step_pair
    await device.connect()
  File "/usr/local/lib/python3.13/site-packages/pyNukiBT/nuki.py", line 389, in connect
    await self._safe_start_notify(
        self._const.BLE_PAIRING_CHAR, self._notification_handler
    )
  File "/usr/local/lib/python3.13/site-packages/pyNukiBT/nuki.py", line 366, in _safe_start_notify
    await self._client.start_notify(*args)
  File "/usr/local/lib/python3.13/site-packages/bleak/__init__.py", line 832, in start_notify
    raise BleakCharacteristicNotFoundError(char_specifier)
bleak.exc.BleakCharacteristicNotFoundError: Characteristic a92ee101-5501-11e4-916c-0800200c9a66 was not found!
```

ronengr, 2025-01-11:
what version of Nuki lock do you have?

polcape, 2025-01-12:
Where can I see it?

ronengr, 2025-01-12:
You don't know what device you have?
It should be written on the package, in the instructions, in the invoice you got, etc.

polcape, 2025-01-13:
It's built in the door, I don't have package but from Nuki app I see Lock type 1.0
[image]

ronengr, 2025-01-13:
I've never tested this with a built-in lock, and never with a 1.0 version. So it makes sense there are some issues.

It seems that the bt char is different. Not sure if it's because of the old lock or because it's built in.

I will try to search for more info about this lock, but if you could find out the bt chars it is using it will be very helpful

polcape, 2025-01-13:
Ok thanks. How can I find them?

polcape, 2025-02-14:
Hi @ronengr,
can you help me?

ronengr, 2025-02-15:
@polcape not sure i can.
I don't have such lock and i didn't see it in the documentation.
If you can find the documentation, or if you research the lock and it's communication and tell me what needs to be modified I will gladly do so.

I think the first step (assuming you don't find any documentation about it) is to find out the bt char device the lock is using.

polcape, 2025-02-16:
Hi @ronengr,

thanks, I understood.
I don't have documentation.

But how can i find the bt char, can you help me to do this?

Thanks

ronengr, 2025-02-17:
while googling it, i found the following thread on stack overflow. maybe it will help you:
https://stackoverflow.com/questions/48638870/how-can-read-all-characteristics-value-of-my-ble-device

testos77, 2025-03-24:
I've been having the same problem with the new **Nuki Smart Lock Go**. Firmware Version is 5.2.2 

` File "/usr/local/lib/python3.13/site-packages/pyNukiBT/nuki.py", line 380, in _safe_start_notify
    await self._client.start_notify(*args)
  File "/usr/local/lib/python3.13/site-packages/bleak/__init__.py", line 832, in start_notify
    raise BleakCharacteristicNotFoundError(char_specifier)
bleak.exc.BleakCharacteristicNotFoundError: Characteristic a92ee101-5501-11e4-916c-0800200c9a66 was not found!
`

MartinKlingeling, 2025-05-08:
I also have the Nuki Smart Lock Go and the same "Characteristic a92ee101-5501-11e4-916c-0800200c9a66 was not found!" error.

Looks like the characteristic starts with a92ee300 and a92ee301 now?

With the app mentioned above, I get this: 

[image]

[image]

[image]

With the bluetoothctl command I connected to the lock and I got this info.

`[bluetooth]# connect 54:D2:72:80:4C:74
Attempting to connect to 54:D2:72:80:4C:74
[CHG] Device 54:D2:72:80:4C:74 Connected: yes
Connection successful
[NEW] Primary Service (Handle 0xefd4)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0001
        00001801-0000-1000-8000-00805f9b34fb
        Generic Attribute Profile
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0001/char0002
        00002a05-0000-1000-8000-00805f9b34fb
        Service Changed
[NEW] Descriptor (Handle 0x9bd0)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0001/char0002/desc0004
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0001/char0005
        00002b2a-0000-1000-8000-00805f9b34fb
        Database Hash
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0001/char0007
        00002b29-0000-1000-8000-00805f9b34fb
        Client Supported Features
[NEW] Primary Service (Handle 0x08c4)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service000e
        0000180a-0000-1000-8000-00805f9b34fb
        Device Information
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service000e/char000f
        00002a25-0000-1000-8000-00805f9b34fb
        Serial Number String
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service000e/char0011
        00002a27-0000-1000-8000-00805f9b34fb
        Hardware Revision String
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service000e/char0013
        00002a26-0000-1000-8000-00805f9b34fb
        Firmware Revision String
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service000e/char0015
        00002a24-0000-1000-8000-00805f9b34fb
        Model Number String
[NEW] Primary Service (Handle 0x1de4)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0017
        a92ee300-5501-11e4-916c-0800200c9a66
        Vendor specific
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0017/char0018
        a92ee301-5501-11e4-916c-0800200c9a66
        Vendor specific
[NEW] Descriptor (Handle 0x9bd0)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0017/char0018/desc001a
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Primary Service (Handle 0x2374)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service001b
        a92ee200-5501-11e4-916c-0800200c9a66
        Vendor specific
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service001b/char001c
        a92ee201-5501-11e4-916c-0800200c9a66
        Vendor specific
[NEW] Descriptor (Handle 0x9bd0)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service001b/char001c/desc001e
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service001b/char001f
        a92ee202-5501-11e4-916c-0800200c9a66
        Vendor specific
[NEW] Descriptor (Handle 0x9bd0)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service001b/char001f/desc0021
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service001b/char0022
        a92ee203-5501-11e4-916c-0800200c9a66
        Vendor specific
[NEW] Primary Service (Handle 0x3ba4)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0024
        0000fff6-0000-1000-8000-00805f9b34fb
        Unknown
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0024/char0025
        18ee2ef5-263d-4559-959f-4f9c429f9d11
        Vendor specific
[NEW] Characteristic (Handle 0x9b88)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0024/char0027
        18ee2ef5-263d-4559-959f-4f9c429f9d12
        Vendor specific
[NEW] Descriptor (Handle 0x9bd0)
        /org/bluez/hci0/dev_54_D2_72_80_4C_74/service0024/char0027/desc0029
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[CHG] Device 54:D2:72:80:4C:74 ServicesResolved: yes
[CHG] Device 54:D2:72:80:4C:74 ServicesResolved: no
[CHG] Device 54:D2:72:80:4C:74 Connected: no
`

I hope this is the info needed to get the lock working.

MartinKlingeling, 2025-05-09:
So I tried a redneck approach and reversed the logic in the connect method in nuki.py to query the "Lock" type with the a92ee301 char first:

[image]

It got me a step further, but new there is a different error: 

```
2025-05-09 01:55:29.208 INFO (MainThread) [pyNukiBT.nuki] Connected successfully
2025-05-09 01:55:29.208 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1,1
2025-05-09 01:55:29.209 INFO (MainThread) [pyNukiBT.nuki] Already connected
2025-05-09 01:55:29.209 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2025-05-09 01:55:29.238 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1,1
2025-05-09 01:55:30.831 INFO (MainThread) [pyNukiBT.nuki] Nuki 54:D2:72:80:4C:74 public key: bb99385fdc0ed5f2f720c8d02df6d01374cd0aa37d1dad73e40ab7819dba5414
2025-05-09 01:55:30.831 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1,1
2025-05-09 01:55:30.832 INFO (MainThread) [pyNukiBT.nuki] Already connected
2025-05-09 01:55:30.832 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2025-05-09 01:55:30.873 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1,1
2025-05-09 01:55:30.876 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1,1
2025-05-09 01:55:30.877 INFO (MainThread) [pyNukiBT.nuki] Already connected
2025-05-09 01:55:30.877 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2025-05-09 01:55:30.948 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1,1

2025-05-09 01:55:32.494 WARNING (MainThread) [pyNukiBT.nuki] Got unexpected message length for command 76. Got 1 unknown bytes with value: b'\x01'
2025-05-09 01:55:32.494 ERROR (MainThread) [pyNukiBT.nuki] HomeAssistant: Received unsolicited notification: Container: 
    command = (enum) (unknown) 76
    payload = None
    unknown = b'\x01' (total 1)
    crc = 46193
2025-05-09 01:55:32.495 ERROR (MainThread) [pyNukiBT.nuki] was expecting CHALLENGE

2025-05-09 01:55:50.949 WARNING (MainThread) [pyNukiBT.nuki] Timeout while waiting for response CHALLENGE on attempt 1
2025-05-09 01:55:50.950 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 2,1
2025-05-09 01:55:50.950 INFO (MainThread) [pyNukiBT.nuki] Already connected
2025-05-09 01:55:50.950 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2025-05-09 01:55:50.973 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 2,1
2025-05-09 01:55:50.975 ERROR (MainThread) [pyNukiBT.nuki] Error K_ERROR_NOT_AUTHORIZED, command EMPTY
2025-05-09 01:55:50.975 ERROR (MainThread) [custom_components.hass_nuki_bt] K_ERROR_NOT_AUTHORIZED
```

ronengr, 2025-06-09:
It seems that not only the char is different but the protocol might be also a bit different.
I couldn't find any documentation about the API for this lock (for the "normal" Nuki they published the API on their web site). Without knowing the API it won't be easy to add support.

ronengr, 2025-09-21:
@polcape @testos77 @MartinKlingeling
The latest version (0.0.19) added support for latest Nuki API to handle gen5 locks.
Could you test to see if it fixes the issue with your lock?