ronengr/hass_nuki_bt #91: Actions are changed and executed even though the state is current.

011V32, 2025-01-13:
### System Health details

## System Information

version | core-2025.1.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | x86_64
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.2
Stage | running
Available Repositories | 1502
Downloaded Repositories | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 468.7 GB
disk_used | 15.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | 

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
[... 132 lines cut ...]
2025-01-13 10:23:36.246 DEBUG (MainThread) [pyNukiBT.nuki] Last action: COMPLETED
2025-01-13 10:23:36.246 DEBUG (MainThread) [pyNukiBT.nuki] HomeAssistant: Fire callbacks
2025-01-13 10:23:36.249 INFO (MainThread) [pyNukiBT.nuki] request_log_entries: COMPLETED
2025-01-13 10:23:36.249 DEBUG (MainThread) [pyNukiBT.nuki] [Container(index=4147, timestamp=datetime.datetime(2025, 1, 13, 9, 23, 28), auth_id=b'\xXX\xXX\xXX\t', name=u'HomeAssistant Oliver', type=uEnumIntegerString.new(2, 'LOCK_ACTION'), data=Container(lock_action=uEnumIntegerString.new(1, 'UNLOCK'), trigger=uEnumIntegerString.new(0, 'SYSTEM'), flags=0, completion_status=uEnumIntegerString.new(0, 'SUCCESS')))]
2025-01-13 10:23:36.250 INFO (MainThread) [pyNukiBT.nuki] sending encrypted command REQUEST_DATA
2025-01-13 10:23:36.250 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1,1
2025-01-13 10:23:36.250 INFO (MainThread) [pyNukiBT.nuki] Already connected
2025-01-13 10:23:36.250 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2025-01-13 10:23:36.296 DEBUG (MainThread) [pyNukiBT.nuki] Got incoming message from: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx (Handle: 32): Unknown
2025-01-13 10:23:36.297 DEBUG (MainThread) [pyNukiBT.nuki] Got command: CHALLENGE
2025-01-13 10:23:36.298 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1,1
2025-01-13 10:23:36.299 INFO (MainThread) [pyNukiBT.nuki] sending encrypted command REQUEST_LOG_ENTRIES
2025-01-13 10:23:36.299 INFO (MainThread) [pyNukiBT.nuki] Trying to send data. Attempt 1,1
2025-01-13 10:23:36.299 INFO (MainThread) [pyNukiBT.nuki] Already connected
2025-01-13 10:23:36.299 INFO (MainThread) [pyNukiBT.nuki] Sending data to Nuki
2025-01-13 10:23:36.353 INFO (MainThread) [pyNukiBT.nuki] Data sent on attempt 1,1
2025-01-13 10:23:36.363 DEBUG (MainThread) [pyNukiBT.nuki] Got incoming message from: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx (Handle: 32): Unknown
2025-01-13 10:23:36.364 DEBUG (MainThread) [pyNukiBT.nuki] Got command: LOG_ENTRY
2025-01-13 10:23:36.394 DEBUG (MainThread) [pyNukiBT.nuki] Got incoming message from: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx (Handle: 32): Unknown
2025-01-13 10:23:36.395 DEBUG (MainThread) [pyNukiBT.nuki] Got command: STATUS
2025-01-13 10:23:36.396 DEBUG (MainThread) [pyNukiBT.nuki] Last action: COMPLETED
2025-01-13 10:23:36.396 DEBUG (MainThread) [pyNukiBT.nuki] HomeAssistant: Fire callbacks
2025-01-13 10:23:36.398 INFO (MainThread) [pyNukiBT.nuki] request_log_entries: COMPLETED
2025-01-13 10:23:36.398 DEBUG (MainThread) [pyNukiBT.nuki] [Container(index=4147, timestamp=datetime.datetime(2025, 1, 13, 9, 23, 28), auth_id=b'\xXX\xXX\xXX\t', name=u'HomeAssistant xxxx', type=uEnumIntegerString.new(2, 'LOCK_ACTION'), data=Container(lock_action=uEnumIntegerString.new(1, 'UNLOCK'), trigger=uEnumIntegerString.new(0, 'SYSTEM'), flags=0, completion_status=uEnumIntegerString.new(0, 'SUCCESS')))]
```

ronengr, 2025-01-13:
As i stated in the other issue, this is by design.
As the Bluetooth connection is not 100% reliable, and sometimes lock/unlock events are missed by ha (very rare if you have good connection, but can happen). I don't trust the status currently stored in ha, and send a lock/unlock command everytime it is requested.

Regarding battery usage, it probably uses less battery than asking the lock for updated state before every locking action.

Regarding the notifications, I'm still thinking about what is the best way to solve it.
I will try to find a solution that won't cause notifications in such case.

One solution i can think of is adding another lock/unlock action that will only preform the action if itvis different from the state currently known to ha.
This is less reliable, but will solve your issue.

011V32, 2025-01-13:
Atm I'm wondering if this isn't a bug in the HomeKit integration.
Because when I switch from Locked to Locking and back to Locked, I first get the message that the lock has been unlocked and then immediately the message that the lock has been locked.
This doesn't actually apply to the two states. Or can it be interpreted that Locking requires Unlocked?