ronengr/hass_nuki_bt #96: Make "lock" the default entity for a device

dotdoom, 2025-02-25:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Currently lock entity is just another one on the device, which gives it a redundant entity id: `lock.{device_name}_lock`.

When `device_name` already contains "lock" (because device is a lock... and I already have more devices associated with the door, such as doorstep light, reed switch sensor etc), this gets even funnier:

[image]

### Describe the solution you'd like

An entity could be set as a "default entity" for the device, which will remove the "Lock" suffix from the name and "_lock" suffix from entity ID, sparing at least one redundant part.

### Describe alternatives you've considered

One can rename entities that already exist on the device (workaround).

### Additional context

Implementation detail: https://developers.home-assistant.io/docs/core/entity/#entity-naming:~:text=The%20entity%20is%20a%20member%20of%20a%20device%20and%20entity.name%20is%20None

dotdoom, 2025-02-25:
This is a trivial change but I'd like to know others' opinions or a discussion.

ronengr, 2025-02-26:
I wasn't aware of the "default entity" possibility.
I don't see any reason not to implement your suggested change.