ronengr/hass_nuki_bt #97: Nuki Ultra: Unable to pair

c00nar, 2025-02-28:
### System Health details

## System Information

version | core-2025.2.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.62-haos-raspi
arch | aarch64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4994
Installed Version | 2.0.5
Stage | running
Available Repositories | 1567
Downloaded Repositories | 21

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.02.4
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 58.0 GB
disk_used | 11.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Studio Code Server (5.18.3), Terminal & SSH (9.16.0), Home Assistant Google Drive Backup (0.112.1), Matter Server (7.0.0), Piper (1.5.2), Whisper (2.4.0), ESPHome Device Builder (2025.2.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
[... 811 lines cut ...]
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 340, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 383, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, user_input["next_step_id"], None
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 490, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/hass_nuki_bt/config_flow.py", line 158, in async_step_pair
    await device.connect()
  File "/usr/local/lib/python3.13/site-packages/pyNukiBT/nuki.py", line 403, in connect
    await self._safe_start_notify(
        self._const.BLE_PAIRING_CHAR, self._notification_handler
    )
  File "/usr/local/lib/python3.13/site-packages/pyNukiBT/nuki.py", line 380, in _safe_start_notify
    await self._client.start_notify(*args)
  File "/usr/local/lib/python3.13/site-packages/bleak/__init__.py", line 832, in start_notify
    raise BleakCharacteristicNotFoundError(char_specifier)
bleak.exc.BleakCharacteristicNotFoundError: Characteristic a92ee101-5501-11e4-916c-0800200c9a66 was not found!
```

oliv3r, 2025-05-30:
According to ChatGPT, the error gets broken down as:

> bleak.exc.BleakCharacteristicNotFoundError: Characteristic a92ee101-5501-11e4-916c-0800200c9a66 was not found!
```
indicates that the Bluetooth characteristic required for pairing with the Nuki lock could not be found by the bleak BLE client library during a start_notify operation.
Breakdown of the Problem

    The component hass_nuki_bt (custom Home Assistant integration for Nuki over Bluetooth) is trying to connect and pair with a Nuki device.

    During this pairing step, it tries to start notifications on the characteristic with UUID a92ee101-5501-11e4-916c-0800200c9a66.

    However, the BLE peripheral (the Nuki device) did not advertise or expose this characteristic.
```

davy-v-2000, 2025-08-13:
Hi there 👋

Just checking in to see if there's been any progress on supporting the Nuki Ultra with this integration? I'm still encountering the same pairing issue as described earlier, and would love to know if there's anything I can test or help with.

Thanks in advance!

ronengr, 2025-08-24:
@davy-v-2000 
I'm looking into it. I hope I will have some progress in the next few days

ronengr, 2025-09-21:
fixed in 0.0.19 thanks to @oliv3r and @YordiDR