ronengr/hass_nuki_bt #112: Show Nuki device as unavailable when no beacon is broadcasted

dotdoom, 2025-07-13:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ronengr/hass_nuki_bt/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

When Nuki lock is unavailable (e.g. I pulled out the battery pack to charge it - but also if BT is somehow broken), the entities retain their last state, which may be confusing or even dangerous - especially if last known state is "locked" and the lock is then externally opened.

### Describe the solution you'd like

Nuki broadcasts a beacon signal every now and then (iirc configurable via API as "normal/slow/slowest"), we could mark integration as failed if no signal received within N seconds.

### Describe alternatives you've considered

- keep everything available with latest state (as is). Probably not a good idea for security?
- only mark unavailable when poll fails.  As it stands, poll will not be scheduled if we didn't receive an advertisement which tells us to poll. So no battery -> no advertisement -> no poll

### Additional context

There's `ActiveBluetoothDataUpdateCoordinator._async_handle_unavailable` overridable callback which can probably be used for this purpose.  At least after pulling the battery out of my Nuki I got a callback within approximately 5 minutes.

I don't know what calls this callback and when, I'll look into this.

ronengr, 2025-07-13:
Interesting idea. I will look into it.

dotdoom, 2025-07-14:
A bit more on context:

- `_async_handle_unavailable` is official API: https://developers.home-assistant.io/docs/core/bluetooth/bluetooth_fetching_data/#activebluetoothcoordinator
- default implementation makes a few changes, including `self._available = False` https://github.com/home-assistant/core/blob/25f64a2f3698b26023bd477efda5f223f7425306/homeassistant/components/bluetooth/update_coordinator.py#L122
- middleware is asking habluetooth to track (un)availability right after start: https://github.com/home-assistant/core/blob/dcbdce4b2b0611439c82e2e8d8cd4743174a682f/homeassistant/components/bluetooth/update_coordinator.py#L99-L106
- the actual implementation is doing some arithmetics to decide when the next advertisement should be, and fires the callback upon missing the deadline: https://github.com/Bluetooth-Devices/habluetooth/blob/21832d237a092ba5e661a992b430d053a80f8b8e/src/habluetooth/manager.py#L392-L414

A fallback deadline value when the logic fails is [15 min](https://github.com/Bluetooth-Devices/habluetooth/blob/21832d237a092ba5e661a992b430d053a80f8b8e/src/habluetooth/const.py#L16-L26).  I think it's reasonable, also given the explanation in that comment.

To mark all entities unavailable one has to set `_available = False` in Coordinator class and update listeners.  This already happens by default:

* `_available=False`: https://github.com/home-assistant/core/blob/25f64a2f3698b26023bd477efda5f223f7425306/homeassistant/components/bluetooth/update_coordinator.py#L115-L122

* `update_listeners`: https://github.com/home-assistant/core/blob/ed4a23d104711e24adfe9133743ca990feaf6556/homeassistant/components/bluetooth/passive_update_coordinator.py#L54-L60

However, the `update_listeners` call does nothing since https://github.com/ronengr/hass_nuki_bt/commit/a0387c09f514da55a61618a252f6c7c81a376120 reimplements add_listener, intercepting parent mechanisms.  `update_listeners` therefore does nothing since that commit, as it's unaware of `_nuki_listeners` field.

@ronengr was that intended?  Could we simply remove `async_add_listener` method override (and accompanying methods/fields) from coordinator.py to restore existing behavior - and automatic "unavailable" state?